NatLabRockies/ROSCO #5: Add a .def file for Windows builds

davidheff, 2020-03-26:
In order to make a Windows DLL with just a single exported function DISCON I think you need to use a .def file.  I included the following file:

`LIBRARY libdiscon.dll

EXPORTS
    DISCON = DISCON`

which does the trick.  I added it to the list of source files in CMakeLists.txt and the subsequent build had the desired export table. But I know absolutely nothing about CMake and probably this is the wrong way to do it!

Having this baked in to the repo would be helpful.

rafmudaf, 2020-03-26:
I thought thats what [this line](https://github.com/NREL/ROSCO/blob/master/src/DISCON.F90#L32) does. How are you looking to see the other symbols exported in the DLL? I'm not too familiar with building software on Windows, but I can try to find a good solution for this. Ideally, these files are autogenerated by CMake or this information is implicit in the code.

davidheff, 2020-03-26:
OK, I may have got this wrong. I'd been building with `-DIMPLICIT_DLLEXPORT` which I had just picked up from

```
if(APPLE OR UNIX)
  if (CMAKE_Fortran_COMPILER_ID STREQUAL "Intel")
    set(CMAKE_Fortran_FLAGS "${CMAKE_Fortran_FLAGS} -DIMPLICIT_DLLEXPORT ")
  else()
    set(CMAKE_Fortran_FLAGS "${CMAKE_Fortran_FLAGS} -DIMPLICIT_DLLEXPORT -ffree-line-length-0")
  endif()
endif()
```

But that results in all symbols being exported, which isn't what you want.

What I will do is look at the openfast `CMakeLists.txt` as you suggested on my other issue and see if I can come up with something definitive.

rafmudaf, 2020-03-26:
Right, my understanding of this was that Windows needs the explicit export preprocessor directive to export any symbols in a DLL. I thought that the default was to export no symbols on Windows, but I may be wrong.

davidheff, 2020-03-26:
Yeah, I've confirmed that my mistake was using `-DIMPLICIT_DLLEXPORT`.

I'm pretty sure that what is needed, at least for an MSYS2 build under mingw-w64 is:

```
set(CMAKE_Fortran_FLAGS "${CMAKE_Fortran_FLAGS} -ffree-line-length-0 -static-libgcc -static-libgfortran -static")
```

closed 2020-04-24 by davidheff