NatLabRockies/ROSCO #11: libdiscon compiled but could not be loaded by OpenFAST on Windows

eolerx, 2020-06-08:
I am trying to compile the ROSCO controller on Windows 10 x64 with the following tools:
- CMake GUI
- MinGW-w64
- the modified CMakeLists.txt file #8 (it would be nice to merge the commit so that other users can access this file from the master branch)

However, I am getting the following error when running the compiled DLL in OpenFAST:
`FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library ..\..\ROSCO\build\libdiscon.dll could not be loaded. Check that the file exists in the specified location and that it is compiled for 64-bit applications.`

Do you know how to resolve this issue ?

FYI, compiling ROSCO on UNIX systems is extremely easy but rather tedious on Windows, even when using Anaconda and the ROSCO toolbox.

eolerx, 2020-06-08:
Resolved -- 32bit instead of 64bit mingw-w64 installed

closed 2020-06-08 by eolerx

si-ctrl, 2023-03-31:
Dear eolerx,

I´ve got the same issue. Would you mind sharing the file?

Thanks

dzalkind, 2023-03-31:
Hi,

Does either of the Windows (dll) binaries here work for you?  https://github.com/NREL/ROSCO/releases/tag/v2.7.0

I agree with the above sentiment regarding Windows vs. Unix compiling, especially for 32-bit.

Best, Dan