NatLabRockies/ROSCO #12: Libdiscon compilation 32bits (BLADED and IVF)

cyril13windglaz, 2020-06-30:
Dear Rosco team,

Thank you so much for your contribution to the open source wind industry knowledged.

I try to compile ROSCO in 32 bits for use with bladed but I didn't managed at the moment.
I try with your Cmake+MINGW32 procedure and also with IVF.
I managed to compile with IVF but then Bladed crash when using it.

Would it be possible that you share a 32bits compile version of your libdiscon.dll ? I could then see if my errors are during the compilation process or when I try to implement the .dll in Bladed.

Best regards,

Cyril

rafmudaf, 2020-07-21:
@cyril13windglaz I _think_ I've compiled this for 32 bit but I dont have a way to verify that. Give it a shot!

[32bit build.zip](https://github.com/NREL/ROSCO/files/4956336/32bit.build.zip)

cyril13windglaz, 2020-07-24:
Hi rafmudaf,

Thx for your try.
Unfortunatly it is not working.

I managed to build one 32bits version with ivf. : 
[libdiscon_32build.zip](https://github.com/NREL/ROSCO/files/4971436/libdiscon_32build.zip)

 I don't know if it is the right way to proceed but for information, below are the modification applied on my side in the .f90.
1. : comment in DISCON.f90 : "avcMSG = TRANSFER(TRIM(ErrMsg)//C_NULL_CHAR, avcMSG, SIZE(avcMSG))"
2. :  change in the subroutine ReadControlParameterFileSub of ReadSetparameters.f90, 
      from    file=accINFILE(1)
      to        file='XXXYOURPATHXX\DISCON.IN',
      in the OPEN command.

3. set the DISCON.IN, libdiscon.dll in the same folder than the Bladed simulation folder
4. copy past in the bladed additionnal controller parameters the content of DISCON.IN
5. select the option "force legacy DISCON, and use floating point protection", time steps 0.01s.

Hope this can help.
If anyone has a better way to proceed please let us know !

Cyril

peter-greaves, 2020-09-07:
Thanks @cyril13windglaz ... This works for me - I would add that the calling convention needs to be __cdecl (C, C++, FORTRAN).

salem-okpokparoro, 2022-10-12:
Below are the steps I took to compile and run ROSCO for Bladed v4.12:
1) Compiling 32bit ROSCO in Windows:
-With same folder structure as ROSCO repository, run the following commands:
cd ROSCO/ROSCO
mkdir build
cd build
cmake .. -G "MinGW Makefiles"   # Windows only
mingw32-make
2) Setting for external controller in Bladed:
>>Control>>Define External Controllers..
Paste DISCON.IN content to the first 'Additional Controller Parameters' box
Set calling convent to _cdecl(C,C++,FORTRAN)
Enable 'Pass Parameters by File?' 
Enable 'Force Legacy DISCON?'
Hope this helps.

Abhi-postdoc, 2022-12-05:
Hi Cyril,
I have followed all the steps mentioned in your post for 32-bit .dll generation for BLADED. However, I got this error msg in the BLADED interface. Please help me to sort this out. Thanks in advance,

*** Note:   Using legacy controller, 'DISCON' in controller C:\Users\PC00\ROSCO\Test_Cases\NREL-5MW\libdiscon.dll
*** ERROR:   Unknown SEH (Structured Exception Handling) error encountered.  This could be due to using the wrong calling convention being used when calling a dll, or a C++ exception being thrown from within an extern "C" function, or a missing dll dependency encountered in external controller.