NatLabRockies/ROSCO #17: S-Function implimentaion

mayankchetan, 2020-07-10:


nikhar-abbas, 2020-07-10:
@mayankchetan you're absolutely right. We're working on getting a distribution of ROSCO in MATLAB-Simulink that matches (as well as possible) the fortran implementation. Will keep this issue open until the distribution exists.

dzalkind, 2020-07-10:
Hey @mayankchetan, I started on a Simulink version of ROSCO here: https://github.com/dzalkind/matlab-toolbox/tree/master/Simulations/SimulinkModels

Its not a perfect recreation or S-Function, but might be helpful.

mayankchetan, 2020-07-10:
@nikhar-abbas , Thank you for the update. 

What would be your opinion on using the ROSCO as a DLL (Setting `PCMode=5` and `VSContrl = 5`) while running OpenFAST as an S-function?

@dzalkind , Thank you! I think the Simulink ROSCO is much better than trying to compile an S-Function. 🙂

nikhar-abbas, 2020-07-10:
@mayankchetan definitely can be done. I've had mixed luck with matlab crashing just trying to directly set `PCMode=5` and `VSContrl = 5` and then running Simulink with the S-function (admittedly, usually an accident), but that should be easy enough to try and see if it works for you. 

Out of curiosity - what would be the use case for this? Just to be able to run ROSCO within your established MATLAB scripts and get the outputs?

mayankchetan, 2020-07-10:
We are looking at implementing an active load controller for WT's. The additional input ports with the OpenFAST S-Function make it easier to prototype.

I might give the DLL ROSCO with OpenFAST S-Function a try and let you know how it goes.

iman-tepsco, 2020-07-14:
Such distribution would be really helpful. I would appreciate it if you can develop the Simulink file for ROSCO.

nikhar-abbas, 2020-07-21:
@mayankchetan, @iman-tepsco 

This has now been wrapped up. A full simulink version of ROSCO is available in the `Matlab_Toolbox` folder on the [develop branch of the ROSCO toolbox](https://github.com/NREL/ROSCO_toolbox/tree/develop). This will get pulled into master in the next week or two, probably

Note that you will need to compile the SFunction, etc., in OpenFAST to get it working. You will also need some tools that are currently @dzalkind's [fork of the matlab-tools](https://github.com/dzalkind/matlab-toolbox) for file handling. We're working on getting these merged into the "official" repo eventually. 

Closing this issue

mayankchetan, 2020-07-21:
@nikhar-abbas, @dzalkind, thank you for the update. I was able to run the ROSCO DLL with the OpenFAST S-Function without issues. Will try the Simulink version now.

iman-tepsco, 2020-07-22:
@nikhar-abbas 
Thank you so much. It will be very helpful.

closed 2020-07-24 by nikhar-abbas