NatLabRockies/ROSCO #21: Build libdiscon 32 bits using MSYS32

peter-greaves, 2020-09-03:
Hi,

I'm trying to create a Bladed version of the IEA 15MW turbine to share on the IEA 15MW repository. I'm really struggling to create a .dll of the controller though - Bladed requires a 32bit version of the DISCON.dll file.

I've managed to successfully compile a libdiscon.so file, but I think Bladed needs a dll (it's throwing an error saying the controller has a divide by zero error). 

What do I need to edit to get the compiler to spit out a dll instead of .so? I can't actually find where the compiler is being called to change the output type.

Thanks,

Pete

rafmudaf, 2020-09-03:
Hello @peter-greaves,

A `.so` is the Linux version of a Windows `.dll`. For what its worth, the macOS version is `.dylib`. These are all generally called "shared libraries". Each is platform specific so you cannot use a `.dll` on mac or a `.so` on Windows. Given that Bladed is loading your `.so`, it looks like you've compiled it correctly. A divide-by-zero-error is a problem, though, so you may want to start thinking about the values given to the DISCON controller and inspect what's happening inside there.

peter-greaves, 2020-09-04:
Thanks @rafmudaf - I've now succesfully compiled to .dll using Code::Blocks (with the changes to the code from the other thread), but I'm still getting the divide by zero error. I think this is more of Bladed problem at this point! If I ever manage to get it working I'll be sure to post my method here.

closed 2020-09-04 by peter-greaves

rafmudaf, 2020-09-04:
Ok so it sounds like you were on a Linux computer or using the WSL on your Windows computer when you initially compiled and that resulted in the `.so` file. And now you've the Code::Blocks IDE to compile the controller directly on Windows instead. Is that correct?

Which Fortran compiler did you use with Code::Blocks? Did you have to install the Intel compiler?

peter-greaves, 2020-09-04:
Yes - I'm on Windows and compiling with MSYS 64/32. I also used this with Code::Blocks - I changed the toolchain and linker paths in Code::Blocks compiler settings to the MYSYS folder. Weirdly this results in a dll when doing it through the same compiler but with the MYSYS terminal gives me a .so!

I tested the compiled files with OpenFAST and they are read in fine, but throw an error to do with reading the DISCON.IN file (I'm not at my computer, but I can remember that the error was at line 69 of the file which reads the input and was to do with a real vs integer problem - I think Fortran automatically assigns types based on where in the alphabet a variable name is?). I posted on the NREL CAE forums about it so full error message is there. I think the best bet is to get it working with OpenFAST and go to Bladed from there, as I get way more feedback about what is going wrong this way.

Thanks a lot for your help!

Peter Greaves

Research Engineer, Blades

Phone: +44 (0) 1670 357741
Mobile:

ore.catapult.org.uk<http://ore.catapult.org.uk>

[https://imagestore.orecat.net/ORE_SM_Colour.jpg]

[https://imagestore.orecat.net/LivingWage-GameChangers.jpg]

The Offshore Renewable Energy Catapult is one of a network of elite technology and innovation centres established by Innovate UK as a long-term investment in the UK’s economic capability. Applying business-led research, Catapults help businesses transform great ideas into valuable products and services to compete in the global markets of tomorrow.

This email is private and confidential and may be legally privileged. If received in error, you must notify us immediately and delete and destroy all copies. The email may contain material protected by intellectual property rights or information exempt from disclosure under freedom of information legislation. You should notify us immediately if asked to disclose information in it in response to a request for information. We make no warranty that this email is error or virus free. Offshore Renewable Energy Catapult is a company limited by guarantee, registered in England and Wales with company number 4659351. Registered office : Offshore House, Albert Street, Blyth, Northumberland, NE24 1LZ.

david-thompson1989, 2021-03-13:
Hi Pete,

I'm trying to get the ROSCO controller working for Bladed too (for the IEA15MW turbine). Did you get it working in the end?

Thanks,
David

peter-greaves, 2021-03-15:
Yes we did - unfortunately we found that when you turn any kind of wave loading on Bladed crashes. We have no idea why unfortunately - we were using it for a study on blade loads so we weren't too worried about this problem. A couple of things we found:

  *   You need to have the full file path to DISCON.IN - a relative path doesn't seem to work (meaning you have to recompile for every project unless you choose to put the controller somewhere suitably generic)
  *   We used Code::Blocks and MinGW - here's a link to our model<https://drive.google.com/file/d/1ffZ-KKGdWl1kdhpXqvanVqhECI8J6VWW/view?usp=sharing>. We found that we had to compile on the target machine (but this may have been before we figured out the wave loading issue) so this may not work for you.
  *   Apart from that I don't think we did anything clever - there's a program called dependency walker which helped a bit too I think.

It was a while ago - sorry if that isn't much help!

Kinfd regards,

Peter

[https://imagestore.orecat.net/ORE_SM_Colour.jpg]

[https://imagestore.orecat.net/LivingWage-GameChangers.jpg]

david-thompson1989, 2021-03-16:
Hi Peter,

Thanks very much for your quick and helpful response. I've now managed to get the controller to work with my Bladed model.

I'll post an update here if I manage to run with wave loading included (I haven't tried yet).

Kind regards,
David

Gawage, 2022-10-10:
HI, 
I have been trying to follow these steps to use ROSCOin 32 Bits with Bladed but have encountered some issues. I have managed to complile a 32 bit .dll with the changes from this thread https://github.com/NREL/ROSCO/issues/12

When trying to use with Bladed I have still encountered errors which Bladed describes as _"This could be due to using the wrong calling convention being used when calling a dll, or a C++ exception being thrown from within an extern "C" function, or a missing dll dependency encountered in external controller."_

I noticed in this thread the it is reccomended to use dependency walker to identify missing depencencies. I have used this program to do this https://github.com/lucasg/Dependencies and it appeards that there are no missing dependencies. Is anyone able to provide more information on identifying missing dependencies? 

I also tried with the new 32 bit libdiscon.dll provided here https://github.com/NREL/ROSCO/releases/tag/v2.6.0 but encountered the same issues. 

Kind Regards 
George