NatLabRockies/ROSCO #25: Estimated wind speed equals to NaN

PE-Meunier, 2020-09-23:
In the develop branch, when using the Kalman wind estimator, the estimated wind speed equals to Nan. 

This behavior does not exist in the master branch.

nikhar-abbas, 2020-09-23:
Hi,
Do you have a little bit more context/details on this - we just pushed some updates to the develop branch that didn't seem to cause WSE instabilities on our end (and actually fixed some problems that were showing up). 

Also, did you test using the most recent master branch? Or an older version? 

Hopefully we didn't introduce any bugs here...

Thanks,
Nikhar

PE-Meunier, 2020-09-24:
Hi,

I did some investigations in the master branch and i discovered that it was working well in "8a5fef005f383a1c45da435ae4283dc1e7c2a1e3" and the issue started in "25a9c319d38c19fb86cbd573416b8fed1c57df86" which corresponds to "Revert "Remove mean wind speed state in EKF"".

I hope it can help you to identify the cause of this issue.

Have a good day,
Paul-Emile

nikhar-abbas, 2020-09-24:
Hi Paul-Emile,
Thanks for diving into this in such detail. In [comparing the the commit](https://github.com/NREL/ROSCO/compare/8a5fef0...25a9c31) that you are saying is stable, 8a5fef005f383a1c45da435ae4283dc1e7c2a1e3, to the recent release, c2662189ba1c09d7b17f2d609ed8ab910dce3b75, I see one or two things that _could_ be causing these issues for you. Unfortunately, I am still unable to re-produce the errors you are seeing on the IEA 15MW Monopile wind turbine. 

1). The turbulence length scale parameter in the wind speed estimator was changes from 3D to 1D. Perhaps setting line 162 in ControllerBlocks.f90 back to `L = 6.0 * CntrPar%WE_BladeRadius` could fix things.  
2) The filtered wind speed estimate used by the torque controller was changed to be the F_LPFCornerFreq defined in the DISCON.IN file when second order filter is used. This is actually a minor bug, but perhaps can be upgraded to a major bug if that is the problem. You can change `CntrPar%F_LPFCornerFreq` to `0.2094` in line 297 so that it reverts to a 30 second time constant for the cutoff frequency. 
3) Finally, the initialization wind speed used to be hard-coded to 10 m/s in line 571 of ReadSetParameters.f90. This was changed to just be the initial wind vane measurement in the recent release, `LocalVar%WE_Vw = LocalVar%HorWindV`.

Otherwise, the main changes in the release were to fix a sign error in the wind speed estimator ( F(2,2) and F(2,3) in the Jacobian), and to move everything to double precision.

Hopefully reverting one of my mentioned changes will help with solving your issues - I am interested to see what it ends up being.

Thanks,
Nikhar

PE-Meunier, 2020-09-25:
Hi Nikhar,

Thanks for your advices. I have managed to find the cause this bug. The divergence of the WSE is removed when reverting back this commit https://github.com/NREL/ROSCO/commit/229c2812d79572c00417d32921808c5b518ebb42.

I have reverted the filtering of the WSE to the second low pass filter with fixed coefficient in the current version (c266218
) of the master branch and it works great. So it seems that the filtering is the cause of the instability (or maybe is the fix of another instability in the WSE).

With the current version of the master branch modified with the second low pass filter with fixed coefficient, I get the best results when using the first order low pass filter configuration (F_LPFType=1) in the discon.in file.

I hope this information will help you.

Have a good day.
Paul-Emile

PE-Meunier, 2020-09-29:
Hi Nikhar,

Instability of the WSE was observed with this set of parameters:
 ```
 # Controller flags
  LoggingLevel:       0                     # {0: write no debug files, 1: write standard output .dbg-file, 2: write standard output .dbg-file and complete avrSWAP-array .dbg2-file
  F_LPFType:          1                     # {1: first-order low-pass filter, 2: second-order low-pass filter}, [rad/s] (currently filters generator speed and pitch control signals)
  F_NotchType:        0                     # Notch filter on generator speed and/or tower fore-aft motion (for floating) {0: disable, 1: generator speed, 2: tower-top fore-aft motion, 3: generator speed and tower-top fore-aft motion}
  IPC_ControlMode:    0                     # Turn Individual Pitch Control (IPC) for fatigue load reductions (pitch contribution) {0: off, 1: 1P reductions, 2: 1P+2P reductions}
  VS_ControlMode:     2                     # Generator torque control mode in above rated conditions {0: constant torque, 1: constant power, 2: TSR tracking PI control}
  PC_ControlMode:     1                     # Blade pitch control mode {0: No pitch, fix to fine pitch, 1: active PI blade pitch control}
  Y_ControlMode:      0                     # Yaw control mode {0: no yaw control, 1: yaw rate control, 2: yaw-by-IPC}
  SS_Mode:            1                     # Setpoint Smoother mode {0: no setpoint smoothing, 1: introduce setpoint smoothing}
  WE_Mode:            2                     # Wind speed estimator mode {0: One-second low pass filtered hub height wind speed, 1: Immersion and Invariance Estimator (Ortega et al.)}
  PS_Mode:            1                     # Pitch saturation mode {0: no pitch saturation, 1: peak shaving, 2: Cp-maximizing pitch saturation, 3: peak shaving and Cp-maximizing pitch saturation}
  SD_Mode:            0                     # Shutdown mode {0: no shutdown procedure, 1: pitch to max pitch at shutdown}
  Fl_Mode:            1                     # Floating specific feedback mode {0: no nacelle velocity feedback, 1: nacelle velocity feedback}
  Flp_Mode:           0                     # Flap control mode {0: no flap control, 1: steady state flap angle, 2: Proportional flap control}
  # Controller parameters 
  zeta_pc:            0.7                   # Pitch controller desired damping ratio [-]
  omega_pc:           0.5                   # Pitch controller desired natural frequency [rad/s]
  zeta_vs:            1.0                   # Torque controller desired damping ratio [-]
  omega_vs:           0.30                 # Torque controller desired natural frequency [rad/s]
```

But was resolved setting:
`F_LPFType:          2                     # {1: first-order low-pass filter, 2: second-order low-pass filter}, [rad/s] (currently filters generator speed and pitch control signals)`

nikhar-abbas, 2020-09-29:
Hi Paul-Emile,

Thanks for these details, they will help. I am working through some tests now and will have some updates pushed shortly.

Nikhar

nikhar-abbas, 2020-10-06:
Hi Paul-Emile,
Sorry about the delay here - there has been a new ROSCO version release with fixes to a bug in one of the filter calls that was causing many of the filters to misbehave.  Sorry about the troubles here!

Going to close #26 as I believe it was the same issues causing problems.

Cheers,
Nikhar

closed 2020-10-06 by nikhar-abbas