NatLabRockies/ROSCO #26: Power optimization issue in develop branch for below nominal wind speed

PE-Meunier, 2020-09-23:
When using the develop branch, I get significantly lower power (GenPwr) than when using the master branch. This comparison has been performed using the same discon file and rotor performance file.

nikhar-abbas, 2020-09-23:
This sounds like it might be related to your previously submitted issue #25.

Can you provide a plot or two showing what's going on here? Hopefully then I can try to re-create the error on my end. 

Also, same question as in #25, are you on the most recent version of the Master branch (v2.1.0), or the previous release (v2.0.3)?

PE-Meunier, 2020-09-24:
I agree with you. The wind estimation error could be the cause of this behavior. Please find below some graph with wind velocity around 7m/s. The black line in the genPower part of the graph represents the expected power at this wind velocity (which was reached with the older version of fast "8a5fef005f383a1c45da435ae4283dc1e7c2a1e3").

[image]

closed 2020-10-06 by nikhar-abbas