NatLabRockies/ROSCO #28: Working with discrepancy results with your publication

Seager1989, 2020-09-29:
Hi，

I compiled the ROSCO on Win10 and tried to replicate your simulation of torque control from your published paper as follows.
"Abbas, Nikhar J., Alan Wright, and Lucy Pao. "An Update to the National Renewable Energy Laboratory Baseline Wind Turbine Controller." Journal of Physics: Conference Series. Vol. 1452. 2020."
The comparison of the results is attached below. I do not know why the generator torque has an immediate dropdown after the velocity change?
From you paper:
[image]
From my simulation:
[image]

Thank you for any comments.

nikhar-abbas, 2020-09-29:
Hi,

This is a good question - thanks for bringing it up!

From a theoretical perspective: the implementation of the torque controller results in a negative DC gain in the transfer function from rotor speed reference to generator torque. This results in the RHP-zero-like behavior that you are wondering about. (Note: The pitch and torque controller tuning parameters define the desired closed loop response of the _rotor speed_, not the actuator signals themselves). We are currently working on a more extensive write-up of the ROSCO controller that will discuss this in more detail.

From an intuitive perspective: ideally, for a step wind change, there is then a proportional step change in the rotor speed reference signal for the TSR tracking PI controller. The PI controller should then reduced the generator torque in order to increase the rotor speed, as it is designed to. In reality though, the step change in wind speed is dependent on the wind speed estimation, which is highly dependent on the rotor's inertial response, numerous filters in the controller, generator torque response, etc. 

There have been many updates to the controller since the time of the initial publication about ROSCO that you are referencing. Notably, some of the filter implementations have changed for the feedback signals that affect the torque controller, which is what you are seeing. Depending on the filter settings and the torque controller tuning parameters, the the RHP can possibly be moved towards the LHP and responses such as those shown in the paper can be seen.

That all said, we have found that the response seen in the step response does not necessarily result in "worse" behavior during turbulent simulations, and that the desired response can be fairly turbine dependent. The specific bandwidth of the controller should  of course be tuned to achieve the desired behavior for the designer. Admittedly, we may need to re-visit the example DISCON.IN files provided in this repo given all of the ROSCO updates. 

Best,
Nikhar

Seager1989, 2020-10-01:
HI Nikhar,

Thank you for your reply. I know more about the ROSCO based on your introduction. I will try to tune it for our model. Thank you

closed 2020-10-06 by nikhar-abbas