NatLabRockies/ROSCO #42: `cmake` is not installed as a dependency

jennirinker, 2021-04-21:
First: thanks for the `setup.py` option to compile the discon controller -- I think it's an extremely useful and quite elegant solution to helpng end-uders update, compile and test ROSCO.

The `cmake` package is not installed with the toolbox, so following the instructions [on the docs](https://rosco-toolbox.readthedocs.io/en/latest/source/install.html#installing-the-rosco-toolbox) will unfortunatley result in an `error: command 'cmake' failed: No such file or directory`.

I'll happily make an MR to put `cmake` as a dependency, but my question is do you want it as a required dependency, or an extra? I would say required makes the most sense, but I defer to you.

nikhar-abbas, 2021-04-21:
Hi Jenni,
Thanks for this! 

I think this is a better suited fix for the ROSCO toolbox as it refers to the `setup.py` compile option. As it is now, the ROSCO toolbox dependencies do not get installed directly during the `python setup.py` call - most dependencies are actually installed with wisdem. 

Could you re-open this issues in the [ROSCO toolbox](https://github.com/NREL/ROSCO_toolbox/issues)? I'll look into better methods of getting cmake automatically installed (or at least have more clear install instructions) soon. For the time being, maybe the open issue will help others.

Going to close this here.

Thanks!
Nikhar

closed 2021-04-21 by nikhar-abbas

jennirinker, 2021-04-22:
Done!
https://github.com/NREL/ROSCO_toolbox/issues/43