NatLabRockies/ROSCO #47: trouble compiling with VStudio

rdamiani, 2021-06-02:
Hello,

I am having difficulties compiling the latest ROSCO with Visual Studio on Windows. I used to be able to compile through the provided .vcxproj without issues. However now I get issues in the custom build step that is using CMake (I believe). 
My CMake executable is under **D:\** but it looks like there is a 'source' that is supposed to be under **C:/**Program Files/[...], which I cannot quite understand. Could you please help me figure this out?
Thank you,

`Build started...
1>------ Build started: Project: ZERO_CHECK, Configuration: Release x64 ------
1>Checking Build System
1>CMake is re-running because D:/[...]/ROSCO/build/CMakeFiles/generate.stamp is out-of-date.
1>  the file '**C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeCommonLanguageInclude.cmake**'
1>  is newer than 'D:/[...]/ROSCO/build/CMakeFiles/generate.stamp.depend'
1>  result='0'
1>-- The Fortran compiler identification is Intel 19.1.2.20200623
1>-- Detecting Fortran compiler ABI info
1>-- Detecting Fortran compiler ABI info - done
1>-- Determine Intel Fortran Compiler Implicit Link Path
1>CUSTOMBUILD : CMake error : The source "D:/Program Files/CMake/share/cmake-3.20/Modules/IntelVSImplicitPath/CMakeLists.txt" does not match the source "C:/Program Files/CMake/share/cmake-3.15/Modules/IntelVSImplicitPath/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
1>CMake Error at D:/Program Files/CMake/share/cmake-3.20/Modules/CMakeDetermineCompilerABI.cmake:155 (try_compile):
1>  Failed to configure test project build system.
1>Call Stack (most recent call first):
1>  D:/Program Files/CMake/share/cmake-3.20/Modules/CMakeTestFortranCompiler.cmake:20 (CMAKE_DETERMINE_COMPILER_ABI)
1>  CMakeLists.txt:2 (project)
1>
1>
1>-- Configuring incomplete, errors occurred!
1>See also "D:/[...]/ROSCO/build/CMakeFiles/CMakeOutput.log".
1>CMake Configure step failed.  Build files cannot be regenerated correctly.  Attempting to stop IDE build.
1>D:\Program Files\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): error MSB8066: Custom build for 'D:\[...]\ROSCO\build\CMakeFiles\5e62dfd0aa8c79d2f8a7b19e54b36a0a\generate.stamp.rule' exited with code 1.
1>Done building project "ZERO_CHECK.vcxproj" -- FAILED.
2>------ Build started: Project: discon, Configuration: Release x64 ------
2>Building Custom Rule D:/[...]/ROSCO/CMakeLists.txt
2>CMake is re-running because D:/[...]/ROSCO/build/CMakeFiles/generate.stamp is out-of-date.
2>  the file 'C:/Program Files/CMake/share/cmake-3.15/Modules/CMakeCommonLanguageInclude.cmake'
2>  is newer than 'D:/[...]/NREL_CODES/ROSCO/build/CMakeFiles/generate.stamp.depend'
2>  result='0'
2>-- The Fortran compiler identification is Intel 19.1.2.20200623
2>-- Detecting Fortran compiler ABI info
2>-- Detecting Fortran compiler ABI info - done
2>-- Determine Intel Fortran Compiler Implicit Link Path
2>CMake Error: The source "D:/Program Files/CMake/share/cmake-3.20/Modules/IntelVSImplicitPath/CMakeLists.txt" does not match the source "C:/Program Files/CMake/share/cmake-3.15/Modules/IntelVSImplicitPath/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
2>CMake Error at D:/Program Files/CMake/share/cmake-3.20/Modules/CMakeDetermineCompilerABI.cmake:155 (try_compile):
2>  Failed to configure test project build system.
2>Call Stack (most recent call first):
2>  D:/Program Files/CMake/share/cmake-3.20/Modules/CMakeTestFortranCompiler.cmake:20 (CMAKE_DETERMINE_COMPILER_ABI)
2>  CMakeLists.txt:2 (project)
2>
2>-- Configuring incomplete, errors occurred!
2>See also "D:/[...]/ROSCO/build/CMakeFiles/CMakeOutput.log".
2>The system cannot find the batch label specified - VCEnd
2>discon : error PRJ0019: A tool returned an error code
2>Build log written to  "file://D:\[...]\ROSCO\build\discon.dir\Release\BuildLog.htm"
2>discon - 1 error(s), 0 warning(s)
========== Build: 0 succeeded, 2 failed, 0 up-to-date, 0 skipped ==========`

rdamiani, 2021-06-02:
I wiped out the build directory, which was corrupted evidently, and went passed that problem, however now I get other issues during compilation: it seems as if the Filters.f90 does not compile due to a type mismatch; any clue on what I should do to bypass this problem? thx

ROSCO\src\Filters.f90(279): error #6633: The type of the actual argument differs from the type of the dummy argument.   [0.]
2>Filters.f90(283): error #6633: The type of the actual argument differs from the type of the dummy argument.   [0.0167]
2>Filters.f90(294): error #6633: The type of the actual argument differs from the type of the dummy argument.   [0.209]
2>Filters.f90(298): error #6633: The type of the actual argument differs from the type of the dummy argument.   [0.7]
2>Filters.f90(299): error #6633: The type of the actual argument differs from the type of the dummy argument.   [0.7]

dzalkind, 2021-06-02:
Hey Rick,

These issues can be compiler-dependent, so they are sometimes hard to know for sure.  In some cases, I've seen that the fortran compiler wants those hard coded filter parameters to have a d0 suffix to be sure it's a double.  

So you'd change 0. to 0.0d0 and 0.0167 to 0.0167d0, etc. in `Filters.f90`. Can you try this and let us know if that works?

rdamiani, 2021-06-02:
Thanks Dan,
indeed I had tried /fpconstant and adding _8 to those numbers. The first option did not make a difference, the second did, however now I have other issues associated with 'saturate' and 'CPfunction'. which, I believe, has the function defined as REAL and may cause another mismatch where 'saturate' requires REAL(8).

rdamiani, 2021-06-02:
Ok So I changed CPFunction to REAL(8), and then added _8 to a number of hardwired numbers throughout the code, and that allowed me to compile. Likely, this is not the best way to solve these issues, and I think using NWTC_Library's ReKi when defining the various vars may be more efficient (?), or perhaps there is another way to bypass the Intel compiler checks on types if REAL(4)/(8) must be used in ROSCO instead. I think there may be a compiler directive for that, but again not sure it is recommended either. Thank you for your time!

closed 2021-06-04 by rdamiani

ghylander, 2021-09-14:
I know this issue is closed, but i too am having trouble with compilation using Visual Studio
Looking at the console log and errors output, it seems the issue is exactly the variable types in Filters.f90 etc

Is there any progress in the fix, or is it being worked on at all?