NatLabRockies/ROSCO #53: WE EKF

rdamiani, 2021-06-30:
Hi Folks,

I have just updated ROSCO to the latest, and am running a controller with, basically, the same input parameters `<>.in` as before. It does not render a reasonable torque control at all. It used to be fine.

After updating ROSCO, there was an issue with the `WE_FOPoles_v` that had the rated wind speed repeated and ROSCO's new `CheckInputs` would complain about non-decreasing values. I then regenerated that list of wind speeds (via the latest toolbox), and that issue went away. Other than that, the new `<>.in` is the same as my previous one except for 
`0.0               ! WE_CP` that is showing one value rather than 3 zeros. I do not think this matters, however. 

Yet the EKF (`2                   ! WE_Mode` )  does not behave anymore. I had it working with constant wind speeds (a stepped ramp) and it used to be accurate, and the torque controller was just fine. Now it is stuck on the initial wind speed (3m/s, just below cut-in). When I set '0                   ! WE_Mode', the controller works as expected.

Could you please help me understand what has changed that triggers this different behavior in EKF?

nikhar-abbas, 2021-06-30:
Hi Rick,
Do you have any way to re-produce this error on our end without sharing proprietary information? Perhaps re-creating it using a reference turbine?

I ran some testing using the IEA15MW floating wind turbine with ROSCO v2.3.0 and OpenFAST v3.0.0. OpenFAST is compiled in single precision using intel compilers, and I tested ROSCO using intel and gcc compilers, along with the ROSCO that gets installed through anaconda using `conda install rosco` (which I then pin as the distributed binaries with each version release). I had no noticeable differences when using these different compilers. 

Running a step wind case for the IEA15-UMaineSemiSum model, I get the following results for initial rotor speeds of 1 and 3 rpms:
[image]
If I set the initial rotor speed to 0 rpm, errors do persist, but this is expected as there is no explicit start-up logic in ROSCO.

As a sanity check, I also looked at the WSE signals and compared them to the rotor averaged wind speed (`RtVAvgxh` in OpenFAST):
[image]

Cheers,
Nikhar

rdamiani, 2021-06-30:
Hi Nikhar,

Thank you for checking this out.
I think I, partially, know what is going on. As I had already mentioned on the toolbox side of things [here](https://github.com/NREL/ROSCO_toolbox/issues/39#issuecomment-868646524) there is a problem with how you guys handled `MinOMSpd`.  There is confusion about what is Rotor and what is Generator RPM, between the toolbox and Rosco. I would  encourage you guys to fix this. 
This may sound a bit harsh, I really appreciate your great work, and this helps us tremendously, just let us not forget the fundamentals :).
I suspect you would see the problem with the NREL 5MW, or a case with Ng<>1, and perhaps with a minimum GenSpd>0.  
Check out lines 246,250,262, `MinOMSpd `should be `MinOMSpd/Ng` for example. I would  check the remainder of the code for possible inconsistencies on Rot vs GenSpd. Also I would check `om_r`, is it properly defined in all instances?
By fixing those lines, I get the controller to work properly if I start above the MinOMSpd threshold (say wind speed of 6 m/s, rather than 3m/s). Yet, If I start from below the MinOMSpd then I encounter the same issue, with EKF is stuck at the initial estimate and the controller does not behave.
Any thoughts?
Thanks again,
Rick

Incorrect behavior:
[image]

This is the correct behavior above threshold:
[image]

nikhar-abbas, 2021-06-30:
Hi Rick,
Yes - this makes sense, thanks for bringing it to our attention again. Sorry about the issue here! It's definitely on the list of changes to be made, just has slipped through the cracks with all the recent updates. I think this bug has flown under the radar for us so for so long because we've been doing a lot of testing with the IEA15MW, which is a direct drive machine. 

I'll get a bug fix pushed here as soon as possible and close this issue when it is resolved.

Cheers,
Nikhar

rdamiani, 2021-06-30:
Thank you Nikhar, do you think there is hope to recover EKF's for GenSpd<= 'MinOMSpd'?

nikhar-abbas, 2021-07-01:
Okay, we are working to combine the ROSCO and ROSCO toolbox git repos into a single repo here: https://github.com/NREL/ROSCO/tree/oneROSCO2. Sorry to throw a wrinkle in yours (well, everyone's) workflows, but this should be easier to manage in the long run. 

I pushed the generator speed/rotor speed fix was made in:
42c6544f08e2c361ffab360d82eb33127d491b6b

As far as getting wind speed estimates for GenSpd <= `MinOmSpd`, I do see your point about not constraining it so that we can still get reasonable estimates below minimum rotor speeds. The constraint was added to avoid the case where rot_speed = 0 and some NaNs are produced, but it does seem better to constrain it to just keep it positive - I made that change in:
874c760c97b2924b7b8ae8abcc84a611868296ef

Going to close this issue now, as I think everything herein has been addressed.

closed 2021-07-01 by nikhar-abbas