NatLabRockies/ROSCO #60: functions type

rdamiani, 2021-07-13:
Hello,

We have witnessed issues with the functions truncating precision as they are declared REAL rather than REAL(8), in particular PIController and Saturate clip the values of GenA/BrTq. This causes issues with VS_State being incorrectly set when saturation/clipping occurs.
I would recommend using ReKi  as for the remainder of OpenFAST, as mentioned earlier on the topic of hard-coded constants, to keep all varsiables and functions consistent; however. I managed to get things to work with REAL(8) in the function declaration.
Best,
Rick

nikhar-abbas, 2021-07-13:
Hi Rick,
Thanks for digging deep into the code and pointing this out. As you might have guessed by the use of REAL(8) instead of REAL(4) in _most_ parts of the code, there is potential for precision errors in a few parts of ROSCO.

As far are shifting everything to be similar to OpenFAST in using ReKi, we have discussed this as a future update. Unfortunately, we just haven't had the bandwidth to make this change yet. 

I'm going to keep this issue open and keep updating it as we move towards a long term solution for things regarding this. 

Cheers,
Nikhar

ghylander, 2021-09-14:
This hardcoding of variable types also makes it impossible to compile ROSCO using Visual Studio and the FORTRAN Intel Compiler

dzalkind, 2021-09-14:
We are addressing this issue in #64, which should make its way into `develop` in the next week or so.

nikhar-abbas, 2022-01-14:
This has been sorted in newer versions of ROSCO, closing this issues.

closed 2022-01-14 by nikhar-abbas