NatLabRockies/ROSCO #62: ROSCO 2.3.0 unstable with IEA15MW turbine

WilliamKeoghDnv, 2021-07-19:
Some recent changes in ROSCO seem to have broken it for use with the IEA15MW model. The control is highly unstable around rated wind speed – continuously oscillating between 6 & 15MW. Example plot below.

We have tried with Bladed, HAWC2 and FAST, and using both ROSCO pre-built dlls downloaded from this github project and building the source ourselves. We see the instability with ROSCO version 2.2.0 & 2.3.0, but not with version 2.0.3. We’re using the same DISCON.IN and Cp_Ct_Cq.IEA15MW.txt text in all cases – exactly what is supplied for IEA15MW in the ROSCO source code package.

Does this work for other people? I can’t see what we could be doing wrong.

Thanks,
Will
[image]

nikhar-abbas, 2021-07-21:
Hi Will,
This is a bit concerning indeed. Unfortunately, I can't seem to replicate your error on my end. Are you simulating for the floating or fixed IEA15MW turbine? 

It looks like you are simulating a fixed bottom turbine based on your shared results. For the sake of simplicity, I have simulated the IEA15MW model [here](https://github.com/NREL/ROSCO_toolbox/tree/main/Test_Cases/IEA-15-240-RWT-UMaineSemi), but with all of the floating DOFs turned off. These results were obtained using OpenFAST v3.0.0 and ROSCO v2.3.0:
[image]

I'm wondering if you have any more information that you can share that might help me replicate your issues? 

Cheers,
Nikhar

WilliamKeoghDnv, 2021-07-22:
Thanks for the reply Nikhar. We're using the non-floating IEA15MW model, from here: https://github.com/IEAWindTask37/IEA-15-240-RWT. Could try with that? 
We were using FAST 8.16, which I gather is slightly different to OpenFAST, so maybe not exactly comparable to what you see. Also the instability we see is significantly less severe with FAST than it is with HAWC2 and Bladed.

nikhar-abbas, 2022-03-25:
This was solved offline by a minor re-tune of ROSCO for the Bladed version of the IEA15MW model - it is distributed with the model [here](https://renewableenergysoftwareportal.dnv.com/KnowledgeBase/Details?productID=1&knowledgeBaseID=212&category=Demo%20models&SearchRawUrl=%2FKnowledgeBase%2FSearch%3FproductID%3D1%26category%3DDemo%2520models).

Thanks for your help with this, @WilliamKeoghDnv + others at DNV!

Closing this issue.

closed 2022-03-25 by nikhar-abbas