NatLabRockies/ROSCO #68: failure to read the Cp_Cq_Ct txt file

ghylander, 2021-09-17:
when using the khalman-filter wind speed estimator, ROSCO terminates with error "forrtl: severe (59): list-directed I/O syntax error, unit 89, file C:\Users\username\Cp_Ct_Cq.NREL5MW.txt"

this also brings me to another issue, ROSCO does not read the file from the directory where the DISCON.IN file is located, isntead it treats my home folder as the cwd

output files however are generated in the correct directory

dzalkind, 2021-09-17:
You could also use an absolute path to the Cp_Ct_Cq file.  We typically run ROSCO using OpenFAST and have the main OpenFAST input file, DISCON.IN file, and Cp_Ct_Cq file all in the same path.

It wouldn't be too difficult to implement some checking on this Cp_Ct_Cq file path, but we need to work through the current PRs first.  We'll keep this open in the meantime.

ghylander, 2021-09-20:
Writing the fulll path works, to directo ROSCO to the file, but the I/O syntax error continues to appear

if the file is not present a "file not found" error is thrown instead

ghylander, 2021-09-20:
For reference, error when the file is not present:

[image]

Error thrown when the file is present:

[image]

dzalkind, 2021-09-20:
When the file is present, that error seems to be about the formatting of the Cp_Ct_Cq file.  Try using the DISCON.IN and Cp_Ct_Cq [here](https://github.com/NREL/ROSCO/tree/oneROSCO2/Test_Cases/NREL-5MW) as a starting point.

ghylander, 2021-09-21:
I did not generate the file using the toolbox, i am already using the Cp_Ct_Cq file from the test files
Either ways, i tried re-downloading both DISCON.IN and the efficiency file from this repo instead of the original toolbox one, same error

ghylander, 2021-09-21:
Could it be an issue with the separator used for terms in the table?
Are you using NEWLINES as value separators? only slashes, commas or blanks are valid value separators

dzalkind, 2021-09-21:
We primarily work with ROSCO on mac/linux, but have had success using cmake on Windows as well.  I would guess that it's related to some compiler/platform specific flags you are using.  Have you tried changing them to separators that are allowed on your platform?

ghylander, 2021-09-22:
I tried removing the newlines from the torque thrust and power coefficient tables but the same list-directed i/o syntax error keeps happening

dzalkind, 2021-09-22:
Is this an issue when you build ROSCO using Visual Studio as described in #66 ?  Is it an issue when you use binaries in our releases or from conda?

ROSCO doesn't seem to have any issues reading other arrays.  Can you please share the Cp_Ct_Cq file you are working with?

Some other debugging ideas:
- You could try figuring out what line ROSCO is failing to read, by printing `i` [here](https://github.com/NREL/ROSCO/blob/7e100154bb8fad0fa10e7bca826d7ee1b68192b1/src/ReadSetParameters.f90#L430)
- You could try reading in the lines as a string, like [here](https://github.com/NREL/ROSCO/blob/7e100154bb8fad0fa10e7bca826d7ee1b68192b1/src/ReadSetParameters.f90#L1255) and printing out what Fortran is trying to read.

ghylander, 2021-09-23:
i was using my mingw32 compiled version of ROSCO, compiled from source without modifications and following the section 4.1 of the ROSCO toolbox developer guide

i now tried using the compiled binaries here, and i got a different error (still using the Cp_Ct_Cq file found in this repo, with no modifications):

[image]

ghylander, 2021-09-23:
After some tinkering around, i got it to load correctly
from my findings, it seems the controlelr i compiled using mingw makefiles and cygwin64 gfortran fails to function properly
i tried other approaches (VS, MSYS), which were succesful at reading the Cp_Ct_Cq file

dzalkind, 2021-09-27:
Hi @ghylander, it sounds like you have this figured out.  Is there anything you would suggest adding here based on what you've learned? https://rosco-toolbox.readthedocs.io/en/latest/source/install.html

ghylander, 2021-09-28:
it seemes to have been caused by a problem with the compilation process.
Confusingly for me, i can not replicate it any more becasue following the exact same compilation procedure now produces a 32 bit dll that can not be executed on my machine.
Any findings i make will be related to #66 so i will close this issue now

closed 2021-09-28 by ghylander

ghylander, 2021-09-30:
I think i've narrowed the issue down, seems to indeed be a compiler issue

it seems that if you compile a dll with a specific compiler, you will then need to execute the dll withthe same compiler runtime, otherwise this issue seems to arise. If i downlaod the lidiscon from releases (which i presume to have been compield with gfrortran) and execute ROSCO with ifort, the controller will have trouble reading the Cp_Ct_Cq file

rafmudaf, 2021-10-01:
@ghylander have you tried compiling and running with MinGW?

There's a nuance where CMake projects using the Intel compiler on Windows don't statically link to runtime libraries. I've spent too much time on this problem with OpenFAST. However, I _think_ the GNU tools do statically link to the runtime libraries.

ghylander, 2021-10-04:
> There's a nuance where CMake projects using the Intel compiler on Windows don't statically link to runtime libraries.

I'm not sure if this is relevant, since running the ifort compiled dll with ifort is ok, the problem is that running a mingw-compiled binary with ifort will result in either NaN wind speed or failure to read the file. I haven't tested yet running ifort-compiled with gfortran because I haven't figured out how to specify which runtime to use and the pc defaults to the intel one.

Right now i have 2 problems with the validation process. On the one hand, I cannot compile ROSCO with ifort then test it on linux with gfortran, since we are not using OpenFAST but an OpenFAST modification called MUST, which is windows-only.

On the other hand, i installed the ROSCO toolbox on my WSL, but when I execute the run_Testing.py file it fails, which is something I'm working on

> have you tried compiling and running with MinGW? 

I'm trying to find a way of running fortran code with a compiler other than the intel one, without uninstalling the latter

Worst-case, you will need to compile the dll with the same compiler you will later run it with