NatLabRockies/ROSCO #69: run test IEA-15-240-RWT-UMaineSemi

robytian, 2021-09-25:
If I want to run this test on Windows, which DLL do I need?
I am currently using the release version of Discon. DLL provided by OpenFast, which can be run. However, the BTS turbulent wind file generated by myself has the following error when running. Is it because of the controller?
[image]
This is my .inp file:
[image]

ghylander, 2021-09-27:
what is your time step?
try reducing it

dzalkind, 2021-09-27:
This controller is not technically the ROSCO controller, so it might be outside of the scope of this repo and might be a better question [here](https://github.com/OpenFAST/r-test) or [here](https://wind.nrel.gov/forum/wind/viewforum.php?f=21&sid=43e59886fb1b6f7f01fe075390e5753c) because I think it has to do with the OpenFAST model.  Like @ghylander said, you might be running with too long of a timestep.  The way I like to debug these issues is to plot the OpenFAST output of the first 80 seconds before the tower strike.  Note that you'll need to set `OutFileFmt` to `1` or `3` in the primary OpenFAST input file to have a text-based output.

closed 2021-09-27 by dzalkind