NatLabRockies/ROSCO #72: oneROSCO2 NREL 5MW DISCON.IN VS_mode is 3

ghylander, 2021-10-01:
however valid values are 0, 1 and 2?

ghylander, 2021-10-01:
also this line needs to be corrected everytime:
`0.00000   1.00000   ! F_FlpCornerFreq`
it will always error our with message 'F_FlpCornerFreq must be greater than zero'
even when Flp_Mode is set to 0 and F_LPFType is set to 1

nikhar-abbas, 2021-10-01:
Hi @ghylander,
Please keep in mind that oneROSCO2 is still in development and not everything has been updated for "production".

VS_mode=3 has been added for TSR-tracking control in below-rated operation, and constant power in above-rated operation. The documentation will be updated accordingly before release.

Nikhar

closed 2021-10-01 by nikhar-abbas

ghylander, 2021-10-04:
> Hi @ghylander, Please keep in mind that oneROSCO2 is still in development and not everything has been updated for "production".
>
> VS_mode=3 has been added for TSR-tracking control in below-rated [...]

That's good to know, but what about the 'F_FlpCornerFreq must be greater than zero' error?
Should it not matter if Flp_Mode is set to 0 and F_LPFType is set to 1?

nikhar-abbas, 2021-10-04:
Sorry about the confusion here - the error catching should be updated so that `Flp_CornerFreq` does not need to be greater than zero unless `Flp_Mode > 0`. This will be fixed before release.