NatLabRockies/ROSCO #74: Documentation errors/bugs

ghylander, 2021-10-05:
# Containment issue for erros/bugs in documentation.
Bugs related to the documentation, it felt like creating individual issues for each one would be bloating the repo

### F_NotchType > 2 is not allowed
Docs say values 0-3 are accepted

### When F_NotchType = 1, ROSCO exits with error
When F_NotchType = 1, Fl_Mode = 1, F_NotchCornerFreq=3.24210, ROSCO exits with error:

`F_NotchType and F_NotchCornerFreq must be specified for Fl_Mode greater than zero.`

This does not happen if F_NotchType = 2.

### PS_Mode > 1 will exit ROSCO
Docs say values 0-3 are accepted

### 'F_FlpCornerFreq must be greater than zero' when FL_Mode = 0
From #72

nikhar-abbas, 2021-10-05:
Thanks for pointing all of these out. Because this is an open-source research code, we have limited resources to keep everything in check (especially these smaller things), so it is good to have this feedback from the users/community to help us out. 

Do you mind putting together a pull request into the `oneROSCO2` branch with these, and any other minor fixes you may see fit? If it is easier, you can combine these updates with whatever solutions you come to as #66 gets resolved.

Thanks,
Nikhar

dzalkind, 2021-10-08:
Thanks for your assistance. All of these are addressed in #54.

In ROSCO, PS_Mode = 1 only enables peak shaving.  In the ROSCO_toolbox, we can peak shave using a few methods.  We can enable PS_Mode of 2 and 3 in ROSCO, but it will still only act as if PS_Mode = 1.  The lookup table is what matters.  @nikhar-abbas, I pushed this change, but I'll let you decide whether to keep it.