NatLabRockies/ROSCO #107: Openfast looking for .so file on mac

ManCla, 2022-01-14:
Dear ROSCO team,

first of all thanks a lot for your work!
I am trying to run your examples on mac but I am having some issues at the interface with openfast. More specifically example_06.py fails with the following output:

```
Using ofTools in ROSCO_toolbox...
-----------------------------------------------------------------------------
Loading wind turbine data for NREL's ROSCO tuning and simulation processeses
-----------------------------------------------------------------------------
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox    
-----------------------------------------------------------------------------
Loading FAST model: IEA-15-240-RWT-UMaineSemi.fst 
Loading rotor performace data from text file: /Users/claudio/repositories-git/wind-turbine-DRC/ROSCO/Test_Cases/IEA-15-240-RWT-UMaineSemi/Cp_Ct_Cq.IEA15MW.txt
Loading rotor performace data from text file: /Users/claudio/repositories-git/wind-turbine-DRC/ROSCO/Examples/../Test_Cases/IEA-15-240-RWT-UMaineSemi/Cp_Ct_Cq.IEA15MW.txt
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox    
-----------------------------------------------------------------------------
Writing new controller parameter file parameter file: /Users/claudio/repositories-git/wind-turbine-DRC/ROSCO/Examples/DISCON.IN.
Using IEA-15-240-RWT-UMaineSemi.fst to run OpenFAST simulation
Running OpenFAST simulation for IEA-15-240-RWT-UMaineSemi.fst through the ROSCO toolbox...

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2021 National Renewable Energy Laboratory
 Copyright (C) 2021 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-HEAD-HASH-NOTFOUND
 Compile Info:
  - Compiler: GCC version 11.1.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Jun 26 2021
  - Time: 17:11:58
 Execution Info:
  - Date: 01/14/2022
  - Time: 18:04:36+0100

 OpenFAST input file heading:
     IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because C_nalpha is 0. (node 2, blade 3)
 Running InflowWind.
 Running ServoDyn.
 Running ServoDyn Interface for Bladed Controllers (using GNU Fortran for Linux, ).

 FAST_InitializeAll:SrvD_Init:BladedInterface_Init:The dynamic library
 ./../../ROSCO/install/lib/libdiscon.so could not be loaded. Check that the file exists in the
 specified location and that it is compiled for 64-bit applications.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

[... 66 lines cut ...]
 Time: 30 of 150 seconds.  Estimated final completion at 18:11:59.                                Generator speed:    6.5 RPM, Pitch angle:   1.2 deg, Power: 10563.5 kW, Est. wind Speed:   9.3 m/s
 Time: 40 of 150 seconds.  Estimated final completion at 18:11:59.                                Generator speed:    6.7 RPM, Pitch angle:   1.0 deg, Power: 10796.7 kW, Est. wind Speed:   9.8 m/s
 Time: 50 of 150 seconds.  Estimated final completion at 18:11:59.                                Generator speed:    6.9 RPM, Pitch angle:   1.7 deg, Power: 11537.7 kW, Est. wind Speed:   9.8 m/s
 Time: 60 of 150 seconds.  Estimated final completion at 18:11:60.                                Generator speed:    7.1 RPM, Pitch angle:   2.2 deg, Power: 12213.9 kW, Est. wind Speed:  10.2 m/s
 Time: 70 of 150 seconds.  Estimated final completion at 18:11:60.                                Generator speed:    7.2 RPM, Pitch angle:   2.6 deg, Power: 12440.2 kW, Est. wind Speed:  10.3 m/s
 Time: 80 of 150 seconds.  Estimated final completion at 18:11:60.                                Generator speed:    7.2 RPM, Pitch angle:   2.6 deg, Power: 12382.5 kW, Est. wind Speed:  10.2 m/s
 Time: 90 of 150 seconds.  Estimated final completion at 18:11:60.                                Generator speed:    7.1 RPM, Pitch angle:   2.2 deg, Power: 12115.4 kW, Est. wind Speed:  10.1 m/s
 Time: 100 of 150 seconds.  Estimated final completion at 18:11:60.                               Generator speed:    7.0 RPM, Pitch angle:   1.8 deg, Power: 11778.1 kW, Est. wind Speed:   9.9 m/s
 Time: 110 of 150 seconds.  Estimated final completion at 18:11:60.                               Generator speed:    6.9 RPM, Pitch angle:   1.2 deg, Power: 11372.0 kW, Est. wind Speed:   9.8 m/s
 Time: 120 of 150 seconds.  Estimated final completion at 18:11:60.                               Generator speed:    6.8 RPM, Pitch angle:   0.7 deg, Power: 11109.5 kW, Est. wind Speed:   9.6 m/s
 Time: 130 of 150 seconds.  Estimated final completion at 18:11:60.                               Generator speed:    6.9 RPM, Pitch angle:   0.6 deg, Power: 11080.0 kW, Est. wind Speed:   9.7 m/s
 Time: 140 of 150 seconds.  Estimated final completion at 18:11:60.                               Generator speed:    6.9 RPM, Pitch angle:   0.7 deg, Power: 11229.4 kW, Est. wind Speed:   9.7 m/s
                                                                                                  
  Total Real Time:       46.223 seconds
  Total CPU Time:        45.32 seconds
  Simulation CPU Time:   42.487 seconds
  Simulated Time:        150 seconds
  Time Ratio (Sim/CPU):  3.5305

  OpenFAST terminated normally.

OpenFAST simulation complete.
```
thanks a lot in advance for the help!
Claudio

rafmudaf, 2022-01-18:
Hi @ManCla

It looks to me like you've resolved the issue in the title (looking for the ".so"), and the problem here is actually in the model inputs with the controller. Is this correct?

Thanks,
Rafael

ManCla, 2022-01-18:
Hi @rafmudaf,
thanks a lot for the reply!

Assuming that my fix is adequate, yes. Now my problem are the warnings and the _fairlead tensions not converging_ (let me remark that I am installing everything now for the first time and I haven't changed anything for now).
If you want I can open a different issue for that?

Thanks again for the response!
Claudio

nikhar-abbas, 2022-01-18:
Hi @ManCla,
Your fix for the .so issue is the correct fix. The CI testing is setup to run on linux systems, so the .so is set as a default. In your case, ROSCO compiles as a .dylib, so you must change the path accordingly. 

As far as the fairlead tensions are concerned, this is the expected behavior for this example. You could change MoorDyn to allot more time for convergence [here](https://github.com/NREL/ROSCO/blob/9646dbdab604471477c40c33023ec776a9ac9ec0/Test_Cases/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_MoorDyn.dat#L31). We tend to ignore the first ~120 seconds or so of a floating simulation to allow for the initial condition transients to die out anyways, so the initial convergence of the fairleads is not of too much concern. 

Hopefully this clarifies things for you!
Nikhar

ManCla, 2022-01-19:
Hi @nikhar-abbas,
thanks a lot for the prompt reply and the clarifications, helped a lot!
Claudio

closed 2022-01-19 by ManCla