NatLabRockies/ROSCO #114: Issue using ROSCO with VS_ControlMode different from 2

ElioDaka, 2022-03-18:
Hello,

I am simulating the behavior of ROSCO on the IEA-15-240-Fixed OpenFast model. When using VS_ControlMode different from 2 combined with the pitch saturation routine (PS_Mode = 1), I obtain negative rotor speed and the Kalman filter cannot estimate the wind speed correctly. I am using uniform laminar wind,  increasing wind speed of 1 m/s every 50 seconds (starting from the cut-in speed). I have tried different initial conditions of the rotor speed but the problem persists. I cannot understand why I get this strange behavior and why everythinf works properly when using VS_ControlMode = 2 with the same initial conditions and keeping constant all the other parameters.
Thank you.
Elio

dzalkind, 2022-03-18:
Hi Elio,

Sounds like this, for our reference: https://forums.nrel.gov/t/pitch-saturation-issue/2956/2

We primarily use VS_ControlMode = 2. Based on the plots in the forum, it looks like the torque command is being saturated at too high of a value.  It may be worth double-checking VS_MinTq, or the pitch saturation logic is changing that value.  I wasn't able to reproduce your issue with [this input file](https://github.com/NREL/ROSCO/files/8306248/DISCON.txt).  Note that we're using the latest develop controller and input file, so you may have to make some changes.

We may have some time to look into this more but can't make any promises.  If you'd like to proceed with your work now, I'd recommend checking out [these lines in the source code](https://github.com/NREL/ROSCO/blob/a857211fa02bd738af54daf6d5a9d1b2cd157265/ROSCO/src/Controllers.f90#L173), starting your studies at a higher wind speed than 3 m/s, or using VS_ControlMode = 2.

If you come up with a solution, please keep us posted, and we would be happy to work with you on incorporating it into our software.

Best, Dan

dzalkind, 2022-03-18:
Hi Elio,

Make sure you're using the latest version of ROSCO.  I think, based on the forum post, that the torque is being saturated to the rated value because there is a pitch above PC_Switch.  When PS_Mode = 1 in the DISCON, the pitch is greater than this value and the torque is saturated to rated.  We made a fix that included the minimum pitch in this calculation.

Best, Dan

ElioDaka, 2022-03-18:
Hi Dan,

I really appreciate your help. I will try to apply your suggestions. When you refer to the "latest version of ROSCO", do you mean the input file DISCON.IN, or do I need to download some new .dll files or something similar so that the fix that you made in the minimum pitch saturation works properly during the OpenFast simulations?
Thank you again.
Best, Elio

dzalkind, 2022-03-18:
Hi Elio,

I mean the controller software/dynamic library, so either [compiling from the source code](https://rosco.readthedocs.io/en/latest/source/install.html#compile-using-cmake) or [downloading the latest release](https://github.com/NREL/ROSCO/releases).

Dan

ElioDaka, 2022-03-22:
Hi Dan,

The problem was that my version of ROSCO was not the latest release, and now all works correclty with all the kind of controllers. Thank you very much!

Elio