NatLabRockies/ROSCO #119: Consult about ROSCO related issues

Francis599, 2022-03-24:
Dear all,
When I run 5MW_Land_Simulink.fst in the file " ROSCO-main\Test_Cases\5MW_Land_Simulink", it often occures fatal errors like "Invalid numerical input" or "Invalid logical input". I dout if the ROSCO_2.4 only refers to OPENFAST_2.4？ Because I use OPENFAST_3.1 but I only find the latest version  ROSCO_2.4 on this web.

nikhar-abbas, 2022-03-25:
@Francis599, the specific OpenFAST version should not matter in this case.

Are you running the model through MATLAB/Simulink, or calling OpenFAST from the command line. If you are calling OpenFAST from the command line, please use runFAST.m from ROSCO's MATLAB toolbox to simulate the turbine instead: 
https://github.com/NREL/ROSCO/tree/main/Matlab_Toolbox

Otherwise, could you please provide more information about the error? It is hard to diagnose with such little information.

dzalkind, 2022-03-28:
Hi @Francis599,

The `main` branch of ROSCO has OpenFAST 3.0.0 Test_Cases/ and the `develop` branch has v3.1.0.  We will be merging `develop` into `main` and releasing sometime this week.  ROSCO does not require any specific version of OpenFAST at this time, but, as Jason said, you need properly formatted input files for the different versions of OpenFAST.

This might be what you are looking for: https://github.com/NREL/ROSCO/tree/develop/Test_Cases/IEA-15-240-RWT-UMaineSemi

Dan

Chueh-Cherng-Jer, 2022-05-18:
Dear all.
When I run the IEA-15-240-RWT-UMaineSemi, it link to the simulink.
But it comes up with the error.
Do you have a solution ?
[image]

dzalkind, 2022-05-18:
Hi,

OpenFAST needs to output the channels that Simulink reads from OutList.  I pushed a small fix here: https://github.com/NREL/ROSCO/pull/135

Chueh-Cherng-Jer, 2022-05-20:
Thank for you answering me!
This question is solved, but there is another problem!
It means that the input port number and output port number is different!
[image]

dzalkind, 2022-05-20:
Hmm, the OpenFAST Sfunction is looking for more inputs than your ROSCO Simulink model is providing it.  I'm not sure I can help you because it depends on your particular OpenFAST version.  Are you using cable or other structural controllers?   I believe I fixed this model so it works with OpenFAST 3.1.0 without any additional inputs, but you can look at the examples [here](https://github.com/OpenFAST/openfast/tree/main/glue-codes/simulink/examples) and tinker with your ROSCO Simulink model near the SFunction so it matches what the OpenFAST example has.

Chueh-Cherng-Jer, 2022-05-21:
Thank you. I will try the solution.
There is another question.
I use the simulation file of IEA 15MW, and run the ROSCO, but it only runs for 7 seconds. 
Do you know what's wrong for my simulation?
[image]

Chueh-Cherng-Jer, 2022-05-22:
Thank for the advise.
I check the simulink file. It shows that the input port number does not match the output port number.
If I modify the number or do other to let it work.
[image]

dzalkind, 2022-05-23:
It seems like this is not an issue in ROSCO, but is more of a discussion of setting up OpenFAST with Simulink.  I would be happy to continue the thread on a discussion page.

closed 2022-05-23 by dzalkind