NatLabRockies/ROSCO #128: Issue in reading Cp_Ct_Cq.txt file

ElioDaka, 2022-04-26:
Dear all,

I am using ROSCO 2.4.1 to perform some simulations using openfast 2.4.0. I have changed the Cp_Ct_Cq.txt file with my own version and I have the following problem: 
"At line 427 of file D:\bld\rosco_1636590340414\work\ROSCO\src\ReadSetParameters.f90 (unit = 89, file = 'C:\Users\Simulation\Files\Cp_Ct_Cq.txt')
Fortran runtime error: Bad real number in item 1 of list input"

Looking at the file ReadSetParameters.f90 at the line 427, it seems that the problem is in the Ct table but i don't have "strange" numbers in that table. What am I missing?
Thank you.
I attache my Cp_Ct_Cq.txt file.
[CpCqCt.txt.txt](https://github.com/NREL/ROSCO/files/8561321/CpCqCt.txt.txt)

ElioDaka, 2022-04-26:
Hi Dan,

Thank you for your help. I checked the line in the DISCON.IN file and I have the right numbers (9 21). As template for the DISCON.IN file and the Cp_Ct_Cq.txt file I used the ones I've found in the .zip file of the ROSCO 2.4.1 release under the IEA-15-240-RWT-UMaineSemi test case folder. Then I updated the file with the model of my turbine. Cp_Ct_Cq.txt file was created simulating the wind turbine without control, imposing pitch angle, TSR and wind speed and obtaining the coefficients. The values of Cp, Ct, and Cq seems reasonable so I don't think the problem is in the numerical values.

Best, Elio

dzalkind, 2022-04-26:
Hi Elio,

Based on the line number, it looks like the Cp table is fine, but it's failing when reading the Ct table.  Perhaps there are some formatting issues there?  Feel free to drop your Cp_Ct_Cq file here along with your DISCON and I'll take a quick look if I have some time.

Best, Dan

ElioDaka, 2022-04-26:
Hi Dan,

Thank you very much. I checked the format and it seems correct to me, but maybe I am missing something.
I drop here the two files.
[Cp_Cq_Ct.txt](https://github.com/NREL/ROSCO/files/8566283/Cp_Cq_Ct.txt)
[DISCON.txt](https://github.com/NREL/ROSCO/files/8566284/DISCON.txt)

Best, Elio

dzalkind, 2022-04-26:
Hi Elio,

Looks like you've found a bug!  The fix on this branch worked for me with your inputs: https://github.com/NREL/ROSCO/tree/hf/table_alloc

We'll release 2.5.1 in the next few days with that fix.  You should be able to use this updated version of ROSCO with OpenFAST 2.4.0, there may be a few lines in the DISCON.IN to update: https://rosco.readthedocs.io/en/latest/source/api_change.html#rosco-v2-4-1-to-rosco-v2-5-0

Thanks for bringing this to our attention!

Best, Dan

ElioDaka, 2022-04-27:
Hi Dan,

Thank you very much! For now I solved the issue using a CpCtCq file with a number of pitch angles greater than the number of TSRs. I will wait for the new release and try again with my inputs. 
Thank you again for your help.

Best, Elio

closed 2022-04-27 by ElioDaka