NatLabRockies/ROSCO #131: ROSCO tuning hyperparameters

C-Giorgio, 2022-05-11:
Hi everyone, 
I am trying to verify from the time response of the turbine the controller parameters that are set in the .yaml file for the pitch controller. I am testing the monopile IEA-15-240-RWT turbine and the desired controller parameters are 1.0 for damping and  0.2 rad/s for the natural frequency (as reported in the .yaml file). To observe this parameters in the rotor speed response, a step input on the reference signal should be applied. Since adding a step input on the reference on the DISCON.IN file is not possible (as far as I know), I run OpenFAST with the Simulink controller. I keep the wind constant at 15 m/s and after 100 s of transient, I apply a step input on the reference generator rotational speed of 0.2 rad/s (red curve). The blue curve is the response of the rotational speed. However, as the figure shows, the system is responding as an underdumped system: large overshoot and few oscillations before reaching the steady-state value. However, since the desired dumping with which the gains are computed  is critical, I would expect a response without oscillations and overshoot. I know that the desired controller parameters are used to compute the gains schedule on a linearized model and what I am testing is a full non-linear model, however I believe the difference is way too large. Is there anything else that I am missing for which this type of analysis is wrong? 

[image]

Any help is really appreciated. Thanks in advance, 
Giorgio

dzalkind, 2022-05-11:
Giorgio,

I'm not entirely sure what you are trying to analyze, but this response looks reasonable.  If you'd like a different response, you can tune the damping (`zeta_pc`) and natural frequency (`omega_pc`).  If you're looking for a more traditional second-order response, I would try a smaller step and avoid the nonlinear near-rated region when blade pitch is near 0.

I don't think this is an issue with ROSCO, so please feel free to pick this up in a discussion post: https://github.com/NREL/ROSCO/discussions 

Best, Dan

closed 2022-05-11 by dzalkind

C-Giorgio, 2022-05-11:
Thank for your response.
What I am trying to analyze is the response of the rotor speed. Since I have impose the damping to be critical (equal to 1.0), I would have expected a response similar to the green one. 
[image]
Anyhow, I didn't believe this is an issue with ROSCO, but more a "my problem".  I will move this  into the discussion section. 

Best, Giorgio