NatLabRockies/ROSCO #144: "Fortran runtime error"  with example_05.py

kercc1985, 2022-06-09:
Hi, Experts:
I am currently learning Rosco, I think I have installed Rosco successfully and run the example in 'standard_use. RST', the 4 examples from 01-04 are running with no problem, but when running 'example_05.py' I get an error like this:

Using ofTools in ROSCO_toolbox...
WARNING: Be sure to pip install simpy and marmot-agents for offshore BOS runs
Loading FAST model: NREL-5MW.fst 
Loading rotor performace data from text file: D:\z各种资料\小工具汇总\ROSCO-main2022-2.5.0\ROSCO-main\Test_Cases\NREL-5MW\Cp_Ct_Cq.NREL5MW.txt
Loading rotor performace data from text file: D:\z各种资料\小工具汇总\ROSCO-main2022-2.5.0\ROSCO-main\Examples\../Tune_Cases\../Test_Cases/NREL-5MW/Cp_Ct_Cq.NREL5MW.txt
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox    
-----------------------------------------------------------------------------
Writing new controller parameter file parameter file: D:\z各种资料\小工具汇总\ROSCO-main2022-2.5.0\ROSCO-main\Examples\DISCON.IN.
Backend TkAgg is interactive backend. Turning interactive mode on.
-------------- Turbine Info --------------
Turbine Name: NREL-5MW
Rated Power: 5000000.0 [W]
Total Inertia: 43702538.1 [kg m^2]
Rotor Radius: 63.0 [m]
Rated Rotor Speed: 1.3 [rad/s]
Max Cp: 0.47
------------------------------------------
                                                                               
------------------------------------------------------------------------------
Running ROSCO-v2.5.0
A wind turbine controller framework for public use in the scientific field    
Developed in collaboration: National Renewable Energy Laboratory              
                            Delft University of Technology, The Netherlands   
------------------------------------------------------------------------------
At line 227 of file C:\Windows\System32\ROSCO\ROSCO\src\ReadSetParameters.f90 (unit = 89)
Fortran runtime error: Cannot open file 'D:\z各种资料\小工具汇总\ROSCO-main2022-2.5.0\ROSCO-main\': No such file or directory

I do not have any clue about this mistake, I hope you can help me find out what is the reason. Thanks.

Regards.

dzalkind, 2022-06-09:
This error is generated because ROSCO can't find your input file.  It seems to think that 'D:\z各种资料\小工具汇总\ROSCO-main2022-2.5.0\ROSCO-main' is the input file, which looks like a directory.  I'm not sure why it's set this way, since the input file is defined [here](https://github.com/NREL/ROSCO/blob/77278f5d1cee95a830b22d79797b351481e87972/Examples/example_05.py#L75).  

I have heard that there are issues with this example on Windows, so perhaps you might skip it.  Otherwise, we need to figure out why the filename is truncated.  [This](https://github.com/NREL/ROSCO/blob/main/ROSCO_toolbox/control_interface.py) is the file that sets up the control interface in python.

kercc1985, 2022-06-09:
Is the input file referred to by this error the 'libdiscon.dll' file?  
I find that 'lib_name' is the input to the 'ControllerInterface' module.
lib_name is given here:
`lib_name= os.path.join(this_dir, '../ROSCO/build/libdiscon.dll')` 

Truncating the string of a path is indeed an odd situation. 
I'll try a little more and see if I can find out the reason.

dzalkind, 2022-06-09:
I think the issue is with `param_file`, perhaps around [here](https://github.com/NREL/ROSCO/blob/77278f5d1cee95a830b22d79797b351481e87972/ROSCO_toolbox/control_interface.py#L100)

kercc1985, 2022-06-10:
dear dzalkind,
You're right. The issue is with `self.accINFILE = self.param_name.encode('utf-8')`. 
When I remove the Chinese path from the project path, the program runs well, although it's not the fundamental solution to the issue.
Thank you again.

closed 2022-06-10 by kercc1985