NatLabRockies/ROSCO #158: Problem met when I run the example files

bkwang18, 2022-07-19:
Hi, 
I need some help for running the example files, when I start to run, it showed this erro:
ModuleNotFoundError: No module named 'wisdem.inputs'
What should I do? I have already installed wisdem in conda install, it shows:
(rosco-env) D:\ROSCO>conda install -y wisdem
Collecting package metadata (current_repodata.json): done
Solving environment: done
# All requested packages already installed.

Thanks.

albertosp3000, 2022-09-28:
Hi @bkwang18! I've got the very same problem. When you say input folder do you mean for instance the example_01.py? Also, what do you mean with Wisdem folder? Is it the one generated in your rosco-env? (e.g. \\anaconda3\envs\rosco-env\Lib\site-packages\wisdem)

I have copied example_01.py in that folder but still not working.

Thanks in advance

dzalkind, 2022-09-28:
Hi, I think you can fix this by doing 

`conda install -y wisdem==3.5.0`

albertosp3000, 2022-09-28:
I've already tried that, and it's not working

albertosp3000, 2022-09-28:
Version 3.5.0:

wisdem                    3.5.0            py39hd4738e6_0    conda-forge

dzalkind, 2022-09-28:
Oh, you might need to re-run `python setup.py develop` in the main ROSCO environment for it to take effect.

When you load up a python terminal and `import wisdem.inputs` it should not throw an error if version 3.5.0 of wisdem is part of the conda environment, which it seems like it is.

albertosp3000, 2022-09-29:
Hi, I've run _python setup.py develop_ in my ROSCO environment, but it still doesn´t work. I've restarted Anaconda Prompt just in case, but it didn´t change anything :-(

albertosp3000, 2022-09-29:
Ok, problem solved. I was using the command  

> py

 to call python. For some reason it was calling a different version. I simply used 

> python

, calling python 3.8, and it works!

dzalkind, 2022-09-29:
Great! I've done that before, too :-)

When I have a lot of different environments, I do `which python` to make sure I'm running the correct one.  That might help with aliases like `py`, too.