NatLabRockies/ROSCO #165: Issue with writing ServoDyn input file via fast-io Rosco-Toolbox

Yuksel-Rudy, 2022-08-20:
Hello developers,
So I was trying to use the reading and writing capabilities of fast-io in ROSCO toolbox. I ran into an issue with the method write_ServoDyn in FAST_writer.py, specifically when writing input data under - STRUCTURAL CONTROL - section. The issue was with writing StC file names. For example:

Intended output for line 67 in ServoDyn input file:
`"unused"      BStCfiles    - Name of the files for blade structural controllers (quoted strings) [unused when NumBStC==0]`
actual output using write_ServoDyn:
`"u" "n" "u" "s" "e" "d"      BStCfiles    - Name of the files for blade structural controllers (quoted strings) [unused when NumBStC==0]`

I fixed the issue in my forked version by changing lines 1296, 1298,1300, and 1302 as follows:

Old:
`f.write('{!s:<22} {:<11} {:}'.format('"' + '" "'.join(self.fst_vt['ServoDyn'] ...`
New:
`f.write('{!s:<22} {:<11} {:}'.format('"' + ''.join(self.fst_vt['ServoDyn'] ...`

To replicate the error using a simple python script:

```
import os
from ROSCO_toolbox.ofTools.fast_io.FAST_reader import InputReader_OpenFAST
from ROSCO_toolbox.ofTools.fast_io.FAST_writer import InputWriter_OpenFAST 

this_dir = os.path.dirname(os.path.abspath(__file__))

fast = InputReader_OpenFAST(FAST_ver='OpenFAST')

fast.FAST_InputFile = 'IEA-15-240-RWT-UMaineSemi.fst'  # FAST input file (ext=.fst)
fast.FAST_directory = os.path.join(this_dir,'../Test_Cases/IEA-15-240-RWT-UMaineSemi') # FAST directory

fast.read_ServoDyn()

fastout = InputWriter_OpenFAST(FAST_ver='OpenFAST')
fastout.fst_vt = fast.fst_vt

fastout.write_ServoDyn()
```

I hope this helps.
Best

dzalkind, 2022-09-12:
Thanks, Rudy!  I have seen this before, too.  Thanks for the nice instructions and sorry for the long delay.  I'll leave this open until we implement your suggested fix.

Dan

Yuksel-Rudy, 2022-09-12:
Thank you Daniel,

I cannot wait to virtually see you tomorrow at AFloat, Session 7.

Best Regards.
Rudy

dzalkind, 2022-09-20:
Hi Rudy,

I implemented this fix in #167 Thanks!

Dan

closed 2022-09-20 by dzalkind