NatLabRockies/ROSCO #166: Full ROSCO installation - --compile-rosco BUG

pablozl92, 2022-09-12:
Hi all,

I have been using ROSCO v2.4.1 for a while, and tried to install the latest version. Following the docs, I have done two little workarounds that may be useful (or require minor code fixing).

The first one has been mentioned in some conversations (#121). 
`conda install -y wisdem` seems to install an older version of wisdem, and `conda install wisdem=3.5.0` seems to fix this.  

The second, as far as I know, is not known.
`python setup.py install --compile-rosco` raises an error, being the most relevant information:

> File "setup.py", line 92, in build_extension                                                                                                                                                                                                   if "gfortran" in os.environ["FC"].lower():                                                                                                                                                                                                 File "C:\Users\Pablo\miniconda3\envs\env_rosco\lib\os.py", line 675, in __getitem__                                                                                                                                                            raise KeyError(key) from None                                                                                                                                                                                                            KeyError: 'FC' 

For me, declaring the env variable inside the conda env to "gfortran" solved the issue
> To set environment variables, run conda env config vars set my_var=value

I hope this is useful!

dzalkind, 2022-09-12:
Thanks!  I'll keep this open until we incorporate these nice tips into our docs.

Dan

dzalkind, 2022-09-12:
Addressed in #167

closed 2022-09-12 by dzalkind

claudius111, 2023-02-15:
> Hi all,
>
> I have been using ROSCO v2.4.1 for a while, and tried to install the latest version. Following the docs, I have done two little workarounds that may be useful (or require minor code [...]

conda env config vars set FC=gfortran