NatLabRockies/ROSCO #171: IEA_10MW_offshore_ROSCO .dll file

wonny0503, 2022-09-19:
Dear ROSCO Community

I am looking for IEA_10MW_offshore_ROSCO .dll file based on 32-bit for DNV Bladed(v.4.10) simulation.

I found **DISCON_IEA_offshore.dll** from IEAWindTask37 / IEA-10.0-198-RWT in Github and it was built with 64-bit.

So I put issues there and couldn't get enough information.

I saw similar issues about compiling IEA_15MW_RWT_Controller based on 32-bit but it's too difficult to understand.

Someone recommended documentation but there isn't detailed information.

Does anyone have ROSCO 2.6.0 for IEA_10MW_offshore? 

If not, Would you please guide me specifically how to build .dll file based on 32-bit, IEA_10MW_offshore? (I have .IN File)

I just stepped into wind industry as a student and don't know how to start.

I really need a information from experts and want to study.

Currently, I am using  DNV Bladed 4.10, OpenFAST 3.2.0, ROSCO 2.6.0, Visual Studio 2019.

Best regards,
wonny

dzalkind, 2022-09-19:
I added a 32-bit dynamic library to the release page: https://github.com/NREL/ROSCO/releases/tag/v2.6.0 

The IEA-10MW DISCON can be found here: https://github.com/IEAWindTask37/IEA-10.0-198-RWT/blob/master/openfast/IEA-10.0-198-RWT_DISCON.IN

wonny0503, 2022-09-20:
Dear dzalkind

I really appreciated your help.

Now I can run Bladed simulation using your 32-bit dynamic library.

There is some errors in result but i guess i can handle this.

If i have another question, i will comment later.

Thanks again and have a nice day!

Best regards,
wonny

closed 2022-09-20 by dzalkind