NatLabRockies/ROSCO #172: Something wrong when using 'runFAST.m'

SYZ622, 2022-09-20:
When I using the 'runFAST.m'  to run openfast_simulink, it appears "WARNING: no outputs found in OutList", and
 "    The expression: u(strmatch('Azimuth',OutList))
    in 'ROSCO/Cyclic Pitch controller/extract Azimuth'
    has a syntax error"
I don't know how to fix this, and I'd like to get your help. Thanks!
Best regards.

dzalkind, 2022-09-20:
`Azimuth` must be outputted from OpenFAST in the ElastoDyn input.

SYZ622, 2022-09-21:
Thanks for your reply! The Azimuth has been outputted in the ElastoDyn, I think there is something wrong in "WARNING: no outputs found in OutList". I don't know why appearing this warning.

SYZ622, 2022-09-21:
Here is my code setting.
clear;

% Compile FAST for use with simulink & mex using openfast docs
fast.FAST_SFuncDir     = 'E:\OpenFAST\openfast_simulink';  %%%% "FAST_SFunc.mexw64, MAP_x64.dll, OpenFAST-Simulink_x64.dll" in this folder
fast.FAST_InputFile    = '5MW_Land_Simulink.fst';   % FAST input file (ext=.fst)
fast.FAST_directory    = 'E:\OpenFAST\ROSCO-main\ROSCO-main\Test_Cases\NREL-5MW';   % Path to fst directory files

% Simulink Parameters
% Model
simu.SimModel           = 'E:\OpenFAST\ROSCO-main\ROSCO-main\Matlab_Toolbox\Simulink\ROSCO';

% Script for loading parameters
simu.ParamScript        = 'E:\OpenFAST\ROSCO-main\ROSCO-main\Matlab_Toolbox\Utilities\load_ROSCO_params';

%% Simulink Setup

[ControlScriptPath,ControScript] = fileparts(simu.ParamScript);
addpath(ControlScriptPath);
addpath(fast.FAST_SFuncDir);
addpath('Utilities')
%% Read FAST Files & Load ROSCO Params from DISCON.IN

[Param,Cx] = ReadWrite_FAST(fast);

% Simulation Parameters
simu.TMax   = Param.FP.Val{contains(Param.FP.Label,'TMax')};
simu.dt   = Param.FP.Val{contains(Param.FP.Label,'DT')};
[R,F] = feval(ControScript,Param,simu);

%% Premake OutList for Simulink

OutList = {'Time'};
OutList = [OutList;
    Param.IWP.OutList;
    Param.EDP.OutList;
    Param.ADP.OutList;
    Param.SvDP.OutList;
    ];

for iOut = 2:length(OutList)
    OutList{iOut} = OutList{iOut}(2:end-1); %strip "s
end

%% Exectute FAST

% Using Simulink/S_Func
FAST_InputFileName = [fast.FAST_directory,filesep,fast.FAST_InputFile];
TMax               = simu.TMax;

SimulinkModel = simu.SimModel;

Out         = sim(SimulinkModel, 'StopTime', num2str(simu.TMax));
sigsOut     = get(Out,'sigsOut');   %internal control signals

%% Get OutData

SFuncOutStr = '.SFunc';

% Try text first, then binary
[OutData,OutList] = ReadFASTtext([fast.FAST_directory,filesep,fast.FAST_InputFile(1:end-4),SFuncOutStr,'.out']);
if isempty(OutData)
    [OutData,OutList] = ReadFASTbinary([fast.FAST_directory,filesep,fast.FAST_InputFile(1:end-4),SFuncOutStr,'.outb']);
end

% Dump data to structure
for i = 1:length(OutList)
    simout.(OutList{i}) = OutData(:,i);
end

%% Plot
Pl_FastPlots(simout)

dzalkind, 2022-09-21:
I would set a breakpoint within `ReadWrite_FAST` and see if the files you're reading exist and if they have an output list.  It should be read from here: https://github.com/NREL/ROSCO/blob/51629665d22f92770f861daeebcebd47ab32abb1/Test_Cases/NREL-5MW/NRELOffshrBsline5MW_Onshore_ElastoDyn.dat#L121

void1981, 2022-09-21:
Same thing here. I think the problem is in the `OutList` itself.

As I run `runFAST.m` for the fst file `ROSCO/Test_Cases/NREL-5MW/NREL-5MW.fst`.
Before `runFAST.m` starts the Simulink file `simulink/ROSCO.mdl`

`Out = sim(SimulinkModel, 'StopTime', num2str(simu.TMax));` line 59 in `runFAST.m`

the `OutList` has only five cells as follows:

```
OutList =

  5×1 cell array

    {'Time'     }
    { 1×1 cell  }
    {36×1 cell  }
    { 1×0 double}
    { 0×1 cell  }

```

This `36x1 cell` contains the following:
```
'"IPDefl1"'
'"TwstDefl1"'
'"BldPitch1"'
'"Azimuth"'
'"RotSpeed"'
'"GenSpeed"'
'"TTDspFA"'
'"TTDspSS"'
'"TTDspTwst"'
'"Spn2MLxb1"'
'"Spn2MLyb1"'
'"RootFxb1"'
'"RootFyb1"'
'"RootFzb1"'
'"RootMxb1"'
'"RootMyb1"'
'"RootMyc1"'
'"RootMyc2"'
'"RootMyc3"'
'"RootMzb1"'
'"RotTorq"'
'"LSSGagMya"'
'"LSSGagMza"'
'"YawBrFxp"'
'"YawBrFyp"'
'"YawBrFzp"'
'"YawBrMxp"'
'"YawBrMyp"'
'"YawBrMzp"'
'"TwrBsFxt"'
'"TwrBsFyt"'
'"TwrBsFzt"'
'"TwrBsMxt"'
'"TwrBsMyt"'
'"TwrBsMzt"'
'"NcIMURAys"'
```
But inside the Simulink file, they call the outputs directly from the `OutList`. For example, calling `Azimuth` as follows:
`u(strmatch('Azimuth',OutList))`
Either rearrange the `OutList` before running the Simulink file `ROSCO.mdl` so that it contains the outputs directly or change the calling function `u(strmatch('Azimuth',OutList))` so that it points to the output which is not suggested I think as you should do for more than 40 output in Simulink file. 

Anyway I guess the whole problem in this code within `runFAST.m` line 46
```
for iOut = 2:length(OutList)
    OutList{iOut} = OutList{iOut}(2:end-1); %strip "s
end
```
Because the `OutList` after this loop is totally different from what it is supposed to be, even the numbers of cells do not match before and after the loop.

`OutList` before the loop is like:
```
OutList =

  5×1 cell array

    {'Time'     }
    { 3×1 cell  }
    {38×1 cell  }
    { 0×0 double}
    { 2×1 cell  }
```
and after the loop:

```
OutList =

  5×1 cell array

    {'Time'     }
    { 1×1 cell  }
    {36×1 cell  }
    { 1×0 double}
    { 0×1 cell  }

```

dzalkind, 2022-09-21:
It also looks like you're pointing to the non-simulink NREL-5MW directory
`fast.FAST_directory = 'E:\OpenFAST\ROSCO-main\ROSCO-main\Test_Cases\NREL-5MW'; % Path to fst directory files`

and the Simulink OpenFAST input file
`fast.FAST_InputFile = '5MW_Land_Simulink.fst'; % FAST input file (ext=.fst)`

I suspect the file readers aren't working properly.

void1981, 2022-09-21:
@dzalkind 

> For reference, here is my `OutList` before the simulation
> 
> ```
> OutList =
> 
>   97×1 cell array
> 
>     {'Time'     }
>     {'Wind1VelX'}
>     {'Wind1VelY'}
>     {'Wind1VelZ'}
>     {'BldPitch1'}
>     {'BldPitch2'}
>     {'BldPitch3'}
>     {'Azimuth'  }
>     {'RotSpeed' }
>     {'GenSpeed' }
>     {'NacYaw'   }
>     {'OoPDefl1' }
>     {'IPDefl1'  }
>     {'TwstDefl1'}
>     {'OoPDefl2' }
>     {'IPDefl2'  }
>     {'TwstDefl2'}
>     {'OoPDefl3' }
>     {'IPDefl3'  }
>     {'TwstDefl3'}
> <lots more>
> ```

is that from `runFAST.m` code?
and which fst file did you use?

dzalkind, 2022-09-21:
https://github.com/NREL/ROSCO/blob/main/Test_Cases/5MW_Land_Simulink/5MW_Land_Simulink.fst

void1981, 2022-09-21:
@dzalkind
Could you share with us your `runFAST.m` file?
It is interesting how would it work for you because I had to change the section `Premake OutList for Simulink` as follows to get rid of this issue.

```
%% Premake OutList for Simulink

OutList= {'Time'};
OutList2 = {'Time'};
OutList2 = [OutList2;
    Param.IWP.OutList;
    Param.EDP.OutList;
    Param.ADP.OutList;
    Param.SvDP.OutList;
    ];

for iOut = 2:length(OutList2)
    OutList = [OutList;OutList2{iOut}(1:end)]; %strip "s
end

for iOut2 = 1:length(OutList)
OutList{iOut2}=strrep(OutList{iOut2},'"','');
end
```
Sure it is not perfect but It works. And `OutList` has only 44 cells when runs [5MW_Land_Simulink.fst](https://github.com/NREL/ROSCO/blob/main/Test_Cases/5MW_Land_Simulink/5MW_Land_Simulink.fst) by default without any modification on any input files (ElastoDyn, ServoDyn, etc).

SYZ622, 2022-09-22:
@void1981 Thanke for your changed code! I'd like to know which version of ROSCO did you use, and could you share with us your runFAST.m file? It still didn't work when I changed the code like your shared. So the "ROSCO.mdl" has no bug for now, right? Thank you!
Best regards,

void1981, 2022-09-22:
@SYZ622 
[ROSCO 2.6.0](https://github.com/NREL/ROSCO/commit/51629665d22f92770f861daeebcebd47ab32abb1)
in `runFAST.m` in section `Premake OutList for Simulink` change this 
```
%% Premake OutList for Simulink

OutList = {'Time'};
OutList = [OutList;
    Param.IWP.OutList;
    Param.EDP.OutList;
    Param.ADP.OutList;
    Param.SvDP.OutList;
    ];

for iOut = 2:length(OutList)
    OutList{iOut} = OutList{iOut}(2:end-1); %strip "s
end
```

to this
```
%% Premake OutList for Simulink

OutList= {'Time'};
OutList2 = {'Time'};
OutList2 = [OutList2;
    Param.IWP.OutList;
    Param.EDP.OutList;
    Param.ADP.OutList;
    Param.SvDP.OutList;
    ];

for iOut = 2:length(OutList2)
    OutList = [OutList;OutList2{iOut}(1:end)]; %strip "s
end

for iOut2 = 1:length(OutList)
OutList{iOut2}=strrep(OutList{iOut2},'"','');
end
```
do you still get the same issue `the expression u(strmatch('Azimuth',OutList)) has a syntax error`?

SYZ622, 2022-09-22:
I have been changed this section. But here is another error.
`Error reported by S-function 'FAST_SFunc' in 'ROSCO/FAST Nonlinear Wind Turbine/S-Function':
FAST_InitializeAll:FAST_Init:FAST_ReadPrimaryFile:Invalid logical input for file
"D:\Users\999\Downloads\ROSCO-2.6.0\ROSCO-2.6.0\Test_Cases\NREL-5MW\5MW_Land_Simulink.fst" occurred while trying to read SumPrint.`

I don't know if my path setting is right. Could you please help me to check? All of path are based on the ROSCO folder.
`clear;

[this_dir,~,~] = fileparts(mfilename('fullpath'));

% Compile FAST for use with simulink & mex using openfast docs
fast.FAST_SFuncDir     = 'D:\Users\999\Downloads\ROSCO-2.6.0\ROSCO-2.6.0';  %%%% NEED FOR SIMULINK
fast.FAST_InputFile    = '5MW_Land_Simulink.fst';   % FAST input file (ext=.fst)
fast.FAST_directory    = 'D:\Users\999\Downloads\ROSCO-2.6.0\ROSCO-2.6.0\Test_Cases\NREL-5MW';   % Path to fst directory files   **(I'm not sure if it's in the "NREL-5MW" folder, if changed to "5MW_Land_Simulink" folder， this code won't read the "Cp_Ct_Cq.NREL5MW.txt", so I copy the  '5MW_Land_Simulink.fst' into "NREL-5MW" folder. )**

% Simulink Parameters
% Model
simu.SimModel           = fullfile(this_dir,'Simulink','ROSCO');

% Script for loading parameters
simu.ParamScript        = fullfile(this_dir,'Utilities','load_ROSCO_params');

%% Simulink Setup

[ControlScriptPath,ControScript] = fileparts(simu.ParamScript);
addpath(ControlScriptPath);
addpath(fast.FAST_SFuncDir);
addpath('Utilities')
%% Read FAST Files & Load ROSCO Params from DISCON.IN

[Param,Cx] = ReadWrite_FAST(fast);

% Simulation Parameters
simu.TMax   = Param.FP.Val{contains(Param.FP.Label,'TMax')};
simu.dt   = Param.FP.Val{contains(Param.FP.Label,'DT')};
[R,F] = feval(ControScript,Param,simu);

%% Premake OutList for Simulink

% OutList = {'Time'};
% OutList = [OutList;
%     Param.IWP.OutList;
%     Param.EDP.OutList;
%     Param.ADP.OutList;
%     Param.SvDP.OutList;
%     ];
% 
% for iOut = 2:length(OutList)
%     OutList{iOut} = OutList{iOut}(2:end-1); %strip "s
% end
%% Premake OutList for Simulink

OutList= {'Time'};
OutList2 = {'Time'};
OutList2 = [OutList2;
    Param.IWP.OutList;
    Param.EDP.OutList;
    Param.ADP.OutList;
    Param.SvDP.OutList;
    ];

for iOut = 2:length(OutList2)
    OutList = [OutList;OutList2{iOut}(1:end)]; %strip "s
end

for iOut2 = 1:length(OutList)
OutList{iOut2}=strrep(OutList{iOut2},'"','');
end

%% Exectute FAST

% Using Simulink/S_Func
FAST_InputFileName = [fast.FAST_directory,filesep,fast.FAST_InputFile];
TMax               = simu.TMax;

SimulinkModel = simu.SimModel;

Out         = sim(SimulinkModel, 'StopTime', num2str(simu.TMax));
sigsOut     = get(Out,'sigsOut');   %internal control signals

%% Get OutData

SFuncOutStr = '.SFunc';

% Try text first, then binary
[OutData,OutList] = ReadFASTtext([fast.FAST_directory,filesep,fast.FAST_InputFile(1:end-4),SFuncOutStr,'.out']);
if isempty(OutData)
    [OutData,OutList] = ReadFASTbinary([fast.FAST_directory,filesep,fast.FAST_InputFile(1:end-4),SFuncOutStr,'.outb']);
end

% Dump data to structure
for i = 1:length(OutList)
    simout.(OutList{i}) = OutData(:,i);
end

%% Plot
Pl_FastPlots(simout)`

dzalkind, 2022-09-22:
I am using the [script](https://github.com/NREL/ROSCO/blob/main/Matlab_Toolbox/runFAST.m) currently available on the main branch and OpenFAST version 3.1.0.

I think you would be better off copying the CpCtCq file into the Simulink directory or using [this branch](https://github.com/dzalkind/ROSCO/tree/vbf), which fixes the relative path issue.

SYZ622, 2022-09-23:
Dear @dzalkind ,
Thanks for your reply!
I have done the copying the CpCtCq file. But there are still two errors.
`Error due to multiple causes.
Caused by:
    The expression: u(strmatch('Azimuth',OutList))
    in 'ROSCO/Cyclic Pitch controller/extract Azimuth'
    has a syntax error
    The expression: u(strmatch('BldPitch1',OutList))
    in 'ROSCO/Extract Signals/Extract BldPitch1'
    has a syntax error`

But when I changed the section of `Premake OutList for Simulink` like @void1981 , another error happened,
`    Error in port widths or dimensions. 'Output Port 1' of 'ROSCO/FAST Nonlinear Wind Turbine/Mux1' has 51 elements.  This port does not accept the
    dimensions (or orientation) specified by the output signal.
    Error in port widths or dimensions. 'Input Port 1' of 'ROSCO/FAST Nonlinear Wind Turbine/S-Function' is a one dimensional vector with 94
    elements.`

closed 2022-10-31 by SYZ622

icnosvbovwb, 2022-11-02:
Dear @void1981 
I run IEA-15-240-RWT-UMaineSemi with ROSCO2.6.0 and use open fast3.1.0. Two errors will appear at the beginning: Error due to multiple causes Caused by: The expression: u(strmatch('Azimuth',OutList)) in 'ROSCO/Cyclic Pitch controller/extract Azimuth' has a syntax error The expression: u(strmatch('BldPitch1',OutList)) in 'ROSCO/Extract Signals/Extract BldPitch1' has a syntax error.  After I modified the code provided by you, the same error occurred: The expression: u (strmatch ('NcIMURAys', OutList)) in 'ROSCO/Extract Signals/Extract NcIMURAys' has a syntax error.  I don't know where there are mistakes. Please help me.
[image]

[image]

void1981, 2022-11-04:
> Dear @void1981 I run IEA-15-240-RWT-UMaineSemi with ROSCO2.6.0 and use open fast3.1.0. Two errors will appear at the beginning: Error due to multiple causes Caused by: The expression: [...]

we have the same issue here. I suggested [this solution](https://github.com/NREL/ROSCO/issues/172#issuecomment-1254516102) to solve these syntax errors. But actually, this is not the main issue. this is caused by something else which is in the Simulink file `ROSCO.mdl`.

If you tried to run `ROSCO.mdl` standalone inside Simulink itself.
It has some unconnected ports and blocks. Also, `FAST_SFunc` block expects different port widths or dimensions. 

```
'Output Port 1' of 'ROSCO/Filter WindSpeed' is not connected.
'Input Port 2' of 'ROSCO/Cyclic Pitch controller/Sum2' is not connected.
Unconnected output line found on 'ROSCO/PitchSaturation' (output port: 2)
Unconnected input line found on 'ROSCO/Cyclic Pitch controller/1P Cyclic Pitch Controller1/1P IPCDQ Filtering1' (input port: 2)

Error in port widths or dimensions. 'Output Port 1' of 'ROSCO/FAST Nonlinear Wind Turbine/Mux1' has 51 elements.  This port does not accept the dimensions (or orientation) specified by the output signal.
Error in port widths or dimensions. 'Input Port 1' of 'ROSCO/FAST Nonlinear Wind Turbine/S-Function' is a one dimensional vector with 94 elements.
```
[image]

what causes these syntax errors is that OpenFAST does not run correctly in the first place. If you looked at the `.out` file you will see it is empty, therefore there are no data to be read.

I think this last error in S-Function is what causes the whole thing. I have tried to modify `ROSCO.mdl` a little but couldn't run it correctly.
I am using .dll file instead.

icnosvbovwb, 2022-11-04:
Dear @void1981 
I found a problem that the X64 bit dll compiled by myself cannot run ROSCO, but the OpenFAST Simulink downloaded from github_ X64.dll can run ROSCO. I'm not sure why.
[image]

SYZ622, 2023-02-24:
> > 亲爱的@void1981我用 ROSCO2.6.0 运行 IEA-15-240-RWT-UMaineSemi 并使用 open fast3.1.0。开头会出现两个错误： Error due to multiple causes Caused by: The expression: u(strmatch('Azimuth',OutList)) in 'ROSCO/Cyclic Pitch controller/extract Azimuth' has a syntax error The expression: u( “ROSCO/Extract Signals/Extract BldPitch1”中的 strmatch('BldPitch1',OutList)) 存在语法错误。我修改你提供的代码后，出现同样的错误： The expression: u (strmatch ('NcIMURAys', OutList)) in 'ROSCO/Extrac

> we have the same issue here. I suggested [this solution](https://github.com/NREL/ROSCO/issues/172#issuecomment-1254516102) to solve these syntax errors. But actually, this is not the main issue. [...]

hi @void1981 
" The expression: u (strmatch ('NcIMURAys', OutList)) in 'ROSCO/Extract Signals/Extract NcIMURAys' has a syntax error"
This error can be solved by changing the variables in the output output of the ELASTODYN file to all be on a separate line. Do not put three outputs in one line. You can try.

lqw075, 2023-05-26:
> hi @void1981 " The expression: u (strmatch ('NcIMURAys', OutList)) in 'ROSCO/Extract Signals/Extract NcIMURAys' has a syntax error" This error can be solved by changing the variables in the output [...]

Could  you tell me exactly how to modify the code?
Thank you,

severalautumn, 2023-07-06:
> 

Dear @void1981 

For the first syntax error with u(strmatch('Azimuth',OutList)), I take this following method:

[https://ww2.mathworks.cn/support/search.html/answers/320615-syntax-error-with-fcn-block-in-simulink.html?fq%5B%5D=asset_type_name:answer&fq%5B%5D=category:simulink/simulink-functions-in-simulink-models&page=1](url)

I am not sure if this is the right way, but it can be running at least currently.

But for your second problem with port widths or dimensions, I also encountered it. Now I'm stuck with this error, did you find any way to resolve it please?

Camille0327, 2025-08-17:
> Could you tell me exactly how to modify the code? Thank you,

Hi, I encountered the same problem with you today. After a long period of debuging, finally I found that once you change the third parameter in "S-Function" to be 0, the problem would be solved.

[image]