NatLabRockies/ROSCO #173: Nacelle yaw controller

MYMahfouz, 2022-09-27:
Hi,

The nacelle yaw controller currently does not account for the platform yaw in case of floating wind turbines.

In the older version with the [ROSCO_toolbox repo](https://github.com/NREL/ROSCO_toolbox). When the floating platform yawed, the nacelle was considered also yawed and the yaw controller was activated to force the nacelle to face the wind.

In the current version, the controller starts working only when the nacelle is yawed relative to the tower reference frame, and not relative to the fixed (inertial) reference frame.
For floating, when this is not considered this means there are cases where there will be constant yaw misalignment.

I was using the ROSCO_toolbox earlier, but I had problems understanding what the fast and slow low pass filters meant, and how to set them.

I hope my issue is clear, and thanks a lot for guiding me to this repo and answering my question earlier.

Best regards,

dzalkind, 2022-09-27:
> In the older version with the [ROSCO_toolbox repo](https://github.com/NREL/ROSCO_toolbox). When the floating platform yawed, the nacelle was considered also yawed and the yaw controller was activated to force the nacelle to face the wind.

I'm not sure this was the case.  That would result in a lot of yaw motion.  Could you show me where this is happening?

In the old version of the yaw controller, here are the default versions of the fast/slow cut-off frequencies: https://github.com/NREL/ROSCO/blob/ad15bcdd6d51e8b537630543d1b97b9048c19d7c/ROSCO_toolbox/utilities.py#L457

Here is the logic explaining the fast/slow yaw error: https://github.com/NREL/ROSCO/blob/ad15bcdd6d51e8b537630543d1b97b9048c19d7c/ROSCO/src/Controllers.f90#L248

MYMahfouz, 2022-09-27:
I have results that I will show here, where I activated the yaw controller.  

I have made the error threshold very small in this simulation. I am aware that this is not realistic. But I also don't know what is the minimum threshold that I can set, I am searching about this in literature.

This simulation shown is a steady state simulation at 10m/s. The wind direction is fixed to 180 degrees. Without the yaw controller my steady state platform yaw is around 8 degrees.

I see that the nacelle is moving a lot and this movement will be much higher once a turbulent wind field is applied. I am not sure how often can the nacelle move and what are the limits, but I am also trying to find out at the moment form the literature. I haven't found anything about this problem for floating wind turbines. Did you see any paper discussing this?

I am using the yaw controller not to totally eliminate the platform yaw but just to decrease its effect. Do you have any other ideas or suggestions to do this? 

Thanks

[image]

dzalkind, 2022-09-27:
We think [this](https://iopscience.iop.org/article/10.1088/1742-6596/1037/3/032011) is a fairly standard yaw controller approach, which is why we implemented it here.  You could compare your controller's results to that.  In other projects, we have used yaw error (which translates to power loss), yaw actuator travel, and the number of yaw activations as measures.  Typically, we want to reduce all those measures, but there is usually a trade-off.  In most yaw controllers, no action is taken if the yaw error is less than 8 degrees because the power loss is not so great and the amount of wind direction variation is larger than that in turbulence.

For floating turbines, mooring lines provide yaw stiffness that reduces platform yaw motion.

MYMahfouz, 2022-09-28:
Thanks a lot for the help. This is really useful.

Yes I agree that the mooring lines provide the yaw stiffness. However, the stiffness vary according to the wind direction and for some wind directions there might be higher yaw values than others, and I am trying to find a solution to decrease the yaw misalignment for these wind directions.

I want to decrease the yaw misalignment not only for the power losses, but also the yaw misalignment creates a crosswind force component on the rotor which also affects the overall floater displacement in the crosswind direction. So I want to decrease this displacement so I aim to decrease the yaw misalignment.

I will compare the controllers as you suggested. 
My last question: are there any plans to update the current version of the yaw controller currently implemented within ROSCO to account for the floater yaw? 

Thanks.

dzalkind, 2022-09-28:
The short answer is yes, the co-dependence of varying yaw stiffness, mooring design, and turbine displacement are on our radar.  The timeline and scope of our ROSCO development efforts are still to be determined.  

However, I think the current controller does account for floater yaw because it still depends on the nacelle vane angle, which should include the platform yaw.  The new controller just includes a [deadband](https://github.com/NREL/ROSCO/blob/51629665d22f92770f861daeebcebd47ab32abb1/Test_Cases/IEA-15-240-RWT-UMaineSemi/DISCON-UMaineSemi.IN#L97) where it won't activate unless the vane angle is greater than some threshold for a certain period of time. 

I'm not sure there are any software issues here, so I'm going to convert this to a discussion to keep it open.