NatLabRockies/ROSCO #183: error on windows trying to kill the dll

rdamiani, 2022-11-18:
`dlclose(handle)` on line 262 of control_interface.py yields this for me when running the ROSCO_walkthrough.ipynb:

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "d:\rrd_engineering\nrel_codes\rosco\ROSCO_toolbox\control_interface.py", line 213, in <lambda>
    dlclose = lambda handle: 0 if _dlclose(handle) else 1
ctypes.ArgumentError: argument 1: <class 'OverflowError'>: int too long to convert`
(note handle=140711689912320).

 Could you please help me fix this ? (it is on windows and based on intel fortran rosco DISCON.DLL)

rdamiani, 2022-11-18:
I found a solution thanks to [this ](https://stackoverflow.com/questions/23522055/error-when-unload-a-64bit-dll-using-ctypes-windll)
I think in the past I had found and fixed another ctype issue with windows dll, which i believe Nikhar had included in one of the previous versions, but this seemed like a different one, and so it could still help others.

dzalkind, 2022-11-21:
Thanks, @rdamiani! Re-opening until I work it into a new PR.

dzalkind, 2022-12-09:
Both this method and the one in there seem to work for me, so I'll keep them both in there.  Added to #188 with [this commit](https://github.com/NREL/ROSCO/pull/188/commits/7035cc3517a5bb3c31eb809d777b77380ed5b109).

closed 2022-12-09 by dzalkind