NatLabRockies/ROSCO #186: example_15 fails due to wrong rosco_dir in run_fast.py

aclerc, 2022-11-24:
I managed to get example_15.py to run but I had to edit line 21 of ROSCO_toolbox/ofTools/case_gen/run_FAST.py

When example_15 runs `r.run_FAST()` then `run_FAST.py` runs:
`python
# Globals
this_dir        = os.path.dirname(os.path.abspath(__file__))
`
The correct answer for `this_dir` should end with `\ROSCO\ROSCO_toolbox\ofTools\case_gen` but for me it ends with `\anaconda3\envs\rosco-env\lib\site-packages\rosco_toolbox\oftools\case_gen\`. I edited line 21 of run_FAST.py in the latter directory so that `this_dir` is correct. I am not sure what a more general fix would be.

dzalkind, 2022-11-28:
Hi @aclerc,

I think you have the \anaconda3\... business because you installed ROSCO without the `develop` in `python setup.py develop`. I'd like this function to work either way, so I'll keep this open until I fix your issue.  Please let me know if you have any ideas in the meantime.  I don't have any at the moment.  You may need to use that `develop` flag to use the function for now.

I have a PR open to fix a bunch of bugs in #188, including one that was breaking the tests in your PRs.  Thanks for bringing this (and the other) issues up!

Best, Dan

dzalkind, 2022-12-09:
I added an attribute to the `run_FAST_ROSCO` class where the `rosco_dir` can be set in the examples: https://github.com/NREL/ROSCO/pull/188/commits/9391f74fe98b428381a636ca6ecdca4cb1c08d32

It seems to be working on my end with your configuration (`python setup.py install`).  It will be merged into `develop` with #188

closed 2022-12-09 by dzalkind