NatLabRockies/ROSCO #191: Below are the steps I took to compile and run ROSCO for Bladed v4.12:

Abhi-postdoc, 2022-12-05:
Below are the steps I took to compile and run ROSCO for Bladed v4.12:
1) Compiling 32bit ROSCO in Windows:
-With same folder structure as ROSCO repository, run the following commands:
cd ROSCO/ROSCO
mkdir build
cd build
cmake .. -G "MinGW Makefiles"   # Windows only
mingw32-make
2) Setting for external controller in Bladed:
>>Control>>Define External Controllers..
Paste DISCON.IN content to the first 'Additional Controller Parameters' box
Set calling convent to _cdecl(C,C++,FORTRAN)
Enable 'Pass Parameters by File?' 
Enable 'Force Legacy DISCON?'
Hope this helps.

_Originally posted by @salem-okpokparoro in https://github.com/NREL/ROSCO/issues/12#issuecomment-1275905293_

dzalkind, 2022-12-06:
When I compiled ROSCO for Bladed, I didn't use mingw32-make.  I had to [install MinGW](https://capsis.cirad.fr/capsis/documentation/mingw-installation) and add it to the PATH. 

The Instructions for Bladed mention something about that _cdecl: https://rosco.readthedocs.io/en/latest/source/ROSCO_instructions_for_Bladed.html

dzalkind, 2022-12-09:
Let's close this one and discuss any necessary changes in #193

closed 2022-12-09 by dzalkind