NatLabRockies/ROSCO #197: Wind turbine curtailment via ROSCO

lorenzoschena, 2022-12-16:
Hello there, 

I was wondering if it is possible to employ ROSCO for power curtailment purposes. 

A first, naive, way to tune the controller to this end might be to modify the `V_rated` of the turbine model to be rated? Is there any drawback that you can see in this approach? 

If yes, can you please point me in the right direction? 

Many thanks,

dzalkind, 2022-12-16:
Hi,

Modifying `v_rated` in the tuning_yaml will not actually help you de-rate.  `v_rated` is used to separate the range of wind speeds used in the automated tuning process.  It's a common misunderstanding about the input we should probably clarify better.

I would change the `rated_rotor_speed` and/or `rated_power`.  Together, those will determine the rated torque, and the ROSCO tuning process should figure out the proper gains and set points from that.

I hope this helps.

Best, Dan

dzalkind, 2023-01-03:
I'm going to convert this to a discussion because it's not really an issue with the code, but perhaps others would like to contribute or learn more.