NatLabRockies/ROSCO #199: Issue with `example_06.py`

lorenzoschena, 2022-12-19:
Hello there, 

I am trying to run `example_06.py`, which tunes a controller and deploys it in an openfast simulation. Unfortunately however, it throws some errors (about input files) when I try to lunch open FAST - the tuning phase runs smoothly. 

The output I get from terminal reads: 

```Loading wind turbine data for NREL's ROSCO tuning and simulation processeses
-----------------------------------------------------------------------------
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox    
-----------------------------------------------------------------------------
Loading FAST model: IEA-15-240-RWT-UMaineSemi.fst 
Loading rotor performace data from text file: /home/ROSCO/Test_Cases/IEA-15-240-RWT-UMaineSemi/Cp_Ct_Cq.IEA15MW.txt
Loading rotor performace data from text file: /home/ROSCO/Examples/../Test_Cases/IEA-15-240-RWT-UMaineSemi/Cp_Ct_Cq.IEA15MW.txt
-----------------------------------------------------------------------------
   Tuning a reference wind turbine controller using NREL's ROSCO toolbox    
-----------------------------------------------------------------------------
Writing new controller parameter file parameter file: /home/ROSCO/Examples/DISCON-UMaineSemi.IN.
Using IEA-15-240-RWT-UMaineSemi.fst to run OpenFAST simulation
Running OpenFAST simulation for IEA-15-240-RWT-UMaineSemi.fst through the ROSCO toolbox...

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2022 National Renewable Energy Laboratory
 Copyright (C) 2022 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST--128-NOTFOUND
 Compile Info:
  - Compiler: GCC version 10.4.0
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Oct 29 2022
  - Time: 12:21:23
 Execution Info:
  - Date: 12/19/2022
  - Time: 11:17:17+0100

 OpenFAST input file heading:
     IEA 15 MW offshore reference model on UMaine VolturnUS-S semi-submersible floating platform

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 3)
 Running InflowWind.
 Running HydroDyn.
   Setting WaveTMax to 0.0 since WaveMod = 0
  WARNING:  The requested output channel is invalid: WavesF1xi
  WARNING:  The requested output channel is invalid: WavesF1zi
  WARNING:  The requested output channel is invalid: WavesM1yi
  WARNING:  The requested output channel is invalid: WavesF2xi
  WARNING:  The requested output channel is invalid: WavesF2zi
  WARNING:  The requested output channel is invalid: WavesM2yi
  WARNING:  The requested output channel is invalid: WavesF2xi
  WARNING:  The requested output channel is invalid: WavesF2yi
  WARNING:  The requested output channel is invalid: WavesF2zi
  WARNING:  The requested output channel is invalid: WavesM2xi
  WARNING:  The requested output channel is invalid: WavesM2yi
  WARNING:  The requested output channel is invalid: WavesM2zi
  WARNING:   The random number generator in use differs from the original code provided by NREL.
  This pRNG uses 8 seeds instead of the 2 in the input file.
  Calculating second order difference frequency wave kinematics.
  Reading in WAMIT output with root name
  "/home/ROSCO/Test_Cases/IEA-15-240-RWT-UMaineSemi/HydroData/IEA-15-240-RWT-UMaineSemi".
  Computing radiation impulse response functions and wave diffraction forces.
  Calculating second order difference-frequency force using the full quadratic transfer function.
 Running MoorDyn (v2.a27, 2022-07-20).
    This is an alpha version of MoorDyn-F v2, with significant input file changes from v1.
    Copyright: (C) 2021 National Renewable Energy Laboratory, (C) 2019 Matt Hall
    Parsing MoorDyn input file:
    /home/ROSCO/Test_Cases/IEA-15-240-RWT-UMaineSemi/IEA-15-240-RWT-UMaineSemi_MoorDyn.dat

 FAST_InitializeAll:HydroDyn_Init:HydroDynInput_ProcessInitData: A requested output channel is
 invalid
 FAST_InitializeAll:MD_Init: Unable to parse Line type 1 on row 7 in input file. Row has wrong
 number of columns. Must be 10 columns.

 OpenFAST encountered an error during module initialization.
  Simulation error level: FATAL ERROR

  Aborting OpenFAST.

Traceback (most recent call last):
  File "/home/ROSCO/Examples/example_06.py", line 87, in <module>
    run_openfast(
  File "/home/miniconda3/envs/fast_env/lib/python3.9/site-packages/ROSCO_toolbox/utilities.py", line 554, in run_openfast
    subprocess.run([fastcall, os.path.join(fastfile)], check=True, cwd=cwd)
  File "/home/miniconda3/envs/fast_env/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['openfast', 'IEA-15-240-RWT-UMaineSemi.fst']' returned non-zero exit status 1.```

Open FAST as a stand-alone works. 
I've followed the installation steps in the documentation, with the only difference that I `conda-forge`-d openfast first then setupping ROSCO. 

I did not change the input files from the example. 

May you please help me? 

Many thanks,

dzalkind, 2022-12-19:
Hi, I'm pretty sure this is because of the [new API in OpenFAST 3.3.0](https://openfast.readthedocs.io/en/main/source/user/api_change.html).  It mostly affects the MoorDyn input.  If you want to run a floating case with OpenFAST through this repo, you can `conda install openfast==3.2.0`, which should work as it did before.

I'll leave this open until we update the inputs of `TestCases/` and the `FAST_Writer.py` for this new MoorDyn input.

lorenzoschena, 2022-12-19:
> Hi, I'm pretty sure this is because of the [new API in OpenFAST 3.3.0](https://openfast.readthedocs.io/en/main/source/user/api_change.html). It mostly affects the MoorDyn input. If you want to run a [...]

As a follow-up, I confirm that downgrading to `OpenFast = 3.2.0` fixed the issue.

Thanks for the prompt response!

dzalkind, 2023-01-03:
OpenFAST 3.3.0 can be used with the `develop` branch of ROSCO now.

closed 2023-01-03 by dzalkind