NatLabRockies/ROSCO #203: Import wisdem error

mftatfcyclone, 2022-12-27:
I installed the widem from conda successfully on Manjaro using
`conda install -y wisdem`

However every time widem library is imported, the following error is generated  
`Python 3.8.15 | packaged by conda-forge | (default, Nov 22 2022, 08:49:35) 
[GCC 10.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import wisdem
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mohamedfekry/WISDEM/wisdem/__init__.py", line 1, in <module>
    from wisdem.glue_code.runWISDEM import run_wisdem
  File "/home/mohamedfekry/WISDEM/wisdem/glue_code/runWISDEM.py", line 16, in <module>
    np.warnings.filterwarnings("ignore", category=np.VisibleDeprecationWarning)
  File "/home/mohamedfekry/.conda/envs/rosco-env/lib/python3.8/site-packages/numpy/__init__.py", line 284, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'warnings'
`

dzalkind, 2022-12-29:
Hi, try reinstalling after downgrading numpy with `conda install numpy=1.23`. The new version of numpy is causing issues with a few of our dependencies.

I'll leave this open because I'd like to figure out how to use 1.24 or make it so folks don't have to manually downgrade.

gbarter, 2022-12-29:
There was a similar message on the WISDEM Issues page.  I have fixed things in this [PR](https://github.com/WISDEM/WISDEM/pull/409) and it has been merged into the WISDEM `develop` branch.  If it is a big need, I can push a minor release and update the conda package as well, but I haven't done that yet.

dzalkind, 2023-02-10:
This has been fixed in #208 and will be released with 2.7.0 soon.

closed 2023-02-10 by dzalkind