NatLabRockies/ROSCO #211: Controller tuning when there is more than 1 WAMIT body

iman-tepsco, 2023-01-20:
I am trying to use ROSCO tuning toolbox for a wind turbine model that has more than 1 WAMIT body, but it fails every time. 
The error is:

```
line 1818, in read_HydroDyn
    self.fst_vt['HydroDyn']['MPropSetID2'][i] = int(ln[4])
IndexError: list index out of range
```

I checked the model, seems there is no problem. 
The OpenFAST model works well without a controller. 

I checked the FAST_reader.py, I understood when there is more than 1 WAMIT body, the toolbox considers NBody for AddF0 (lines 1634-1640).
But it seems AddCLin,AddBLin, and AddBQuad are not updated and the toolbox considers a 6*6 matrix (lines 1642-1644), while I think the size of these matrices should be (NBody*6)*(NBody*6).

I am new to both hydrodynamics and Python programming, so I am not sure if my understanding is correct. 
Please let me know if my understand is correct and if so, please let me know what changes should I apply. 
If my understanding is not correct, please let me know what you think about the above-mentioned error. 

Thank you so much.

dzalkind, 2023-01-23:
Hi,

Based on the error, it looks like it's failing when reading the members and not when setting up the added parameters for the WAMIT bodies.  I'm not sure of the exact details required for `MPropSetID2` and the other parameters required under `MEMBERS` in HydroDyn.  If it works without the controller tuning, what version of OpenFAST does it work with?  There could be a version mismatch between your model and the `FASTReader`.

Not much of HydroDyn is actually required for tuning the controller.  You could work around this by commenting out [these lines](https://github.com/NREL/ROSCO/blob/05d7b3b948c12ad40892941b6f92f3b08f5c6894/ROSCO_toolbox/turbine.py#L177).  If you want to run batch simulations using ROSCO, you'll need to figure out how to read those inputs, so they can be written back out for batch runs.

Best, Dan

dzalkind, 2023-02-10:
Hi,

Since I haven't heard any updates, I'll close this for now.  Please re-open if you run into further issues.  The upcoming release will be compatible with OpenFAST 3.4.0.

Best, Dan

closed 2023-02-10 by dzalkind