NatLabRockies/ROSCO #219: vs_rgn2K calculation in Rosco Toolbox

BenoitSchiltz, 2023-02-10:
Hello,

I am shifting from using the Rosco toolbox made for the ROSCO 2.4 to the one made for ROSCO 2.6, and I noticed a change in the calculation of K for Komega_g^2 (vs_rgn2K) in controller.py between the two versions, with the the removal of the generator efficiency consideration in the version 2.6 compared to the 2.4, where it was used. I did not find any change to how Controller.f90 deals with vs_rgn2K between the two versions though. 

Could you explain which version is correct, or if I missed something in the source code or in the log change from version 2.4 to 2.6? 

Thanks
Best regards

dzalkind, 2023-02-10:
Hi,

I would assume the newer version is correct.  I don't recall the exact reason for the change, but I would guess that for some reason, both `GenEff` and `GBEff` do not need to be included.  In most cases, the difference will be very small.  If you find that there should be a change, please let me know. 

Best, Dan

BenoitSchiltz, 2023-02-10:
Thanks for the quick reply. Meanwhile I dug a bit more in the code and understood that the k coefficient is not used when optimal TSR tracking is selected, which is what I use. 

Benoît

closed 2023-02-10 by dzalkind