NatLabRockies/ROSCO #220: Blade pitch oscillation near rated wind speed (onshore steady)

BenoitSchiltz, 2023-02-10:
Hi all,

After picking the damping and frequency (for pitch control) that would minimise tower fatigue on a floating wind turbine with turbine derived from the IEA 15 MW, I get blade pitch oscillations (thus power and rotor speed) near rated wind speed in steady and onshore conditions.

I was not getting that result when I was using the original coefficients for UMaine (zeta 1 omg 0.2 rad.s-1). Do you know why the signal does not converge? My controller uses different peak shaving and operational points than the UMaine, and the damping and frequency for blade pitch control are 1.6 and 0.22 rad.s-1.

Here is a picture of my blade pitch signal with wind speed slightly above rated wind speed.

[image]

Thanks
Best regards
Benoît

dzalkind, 2023-02-10:
Hi Benoit,

Often, the best tuning parameters for one turbine are not the best for all turbines. You can fix the issue you see by decreasing `omega_pc` or increasing `ps_percent`.  In some cases, the best controller in turbulence simulations will oscillate in a steady wind at rated wind speed; it is actually a tricky case.

There is also an option to decrease the `omega_pc` near-rated wind speeds only.  Please see this example: https://github.com/NREL/ROSCO/blob/51629665d22f92770f861daeebcebd47ab32abb1/Tune_Cases/IEA15MW_MultiOmega.yaml#L42 

Best, Dan

BenoitSchiltz, 2023-02-21:
Hello Dan, 

Thanks for your answer. I found other controller parameters that also optimise my fatigue without displaying oscillation around rated wind speed using steady wind. However, if I run into a similar problem as described above again, could you explain the reason why for some controller parameters minimizing fatigue in turbulent wind, there are controller oscillations around rated wind speed in steady wind ? 
And to some extent, can a controller design be valid only based on turbulent simulations and not steady wind simulations if such oscillations appear?

Concerning the optimization of the controller, could you explain why you chose in ROSCO toolbox (and by extent ROSCO) to define several breakpoints `omega_pc`, `zeta_pc` wind speed wise, rather than `Fl_Kp` breakpoints wind speed wise? 
This would help define the best controller in terms of `omega_pc` and `zeta_pc` and then tune `Fl_Kp` wind speed wise, no? 
I guess allowing breakpoints on two parameters instead gives more flexibility and better results, even though optimization problem is harder. Could you comment on that?

From your paper "Floating wind turbine control optimization", picture 3 on frequency and damping plots, is it true to say that from the optimal `Fl_Kp`, for fatigue optimization the controller should be slower around rated wind than at higher wind speeds when using multiple breakpoints? If yes, why is that?

Best regards
Benoît

dzalkind, 2023-02-21:
Hi Benoit,

The reason for instability near rated is the typical negative or low damping problem associated with floating wind turbines.  It's worst near-rated, and a small amount of turbulence can move the operating point away from wind speeds with less damping, somewhat avoiding the problem.

I chose to optimize `omega_` and `zeta_pc` w.r.t. wind speed because those are the primary drivers of the negative damping issue.  However, gain scheduling `Fl_Kp` is something we also believe is useful and is on our TODO list.  So, I think you are correct.  If you find a nice solution and would like to contribute, please feel free to open a PR.  Technically, within the ROSCO_toolbox, it is gain scheduled, but we choose the gain corresponding to the rated wind speed.  I don't necessarily think applying that gain scheduling directly is the best solution, for reasons I'll explain below.

In optimizations, I have seen that a more aggressive (less negative) `Fl_Kp` performs better for damping platform motion, especially near-rated.  At higher wind speeds, the thrust on the rotor is less, and the coupling with the generator speed is also less, so a lower magnitude of `Fl_Kp` is needed.  Also, waves are typically greater at higher wind speeds, and we have seen that the floating feedback in ROSCO can amplify (or at least doesn't dampen) platform motion due to waves.  So, that is another reason to decrease `Fl_Kp` magnitude at high wind speeds.

I hope this helps.

Best, Dan

BenoitSchiltz, 2023-02-21:
Dan,

Thanks for the input. Sorry it was not clear, but in my first post, I was describing a tuned controller fatigue wise on floating, that once applied on an onshore wind turbine behaves baddly around rated wind speed. If you have some thoughts about that, I am interested. I'll check the behavior in floating steady wind conditions as well.

Ok, so just to be sure I understand correctly, your 3 points suggest `Fl_Kp` value closer to 0 is better for platform motion and thus for fatigue, is that right? I have also noticed a validity domain of floating feedback gain when looking for optimum parameters. 
However, when doing a floating feedback sensitivity (not changing `omega_pc` and `zeta_pc`), I found that the floating feedback gain can be more negative than what ROSCO_toolbox calculates for all wind speed in order to lower fatigue on the tower without affecting much the mean power.  I do see that for constant damping and frequency, the optimum `Fl_Kp` reduces with wind speed.

In your paper I mentionned above, you also found quite negative floating feedback gain to optimise fatigue.
Then I am not sure how to understand your sentences wrt the paper and my findings. Have you found something different since publishing the paper?

Thanks again
Benoît

dzalkind, 2023-02-21:
Hi Beniot,

All turbines are a little unstable near rated.  Are you using peak shaving in the fixed-bottom controller?  That may be what causes the instability. Peak shaving also change the operating point away from unstable operating points.

I am suggesting that an `Fl_Kp` with a greater magnitude, like -20 or more is better for platform motion, like the paper.  At high wind speeds and in large waves, I think something closer to 0 is better.

Best, Dan

dzalkind, 2023-03-17:
It sounds like we've addressed any issues you're having.  I'll convert this to a discussion to keep the conversation going.

closed 2023-03-17 by dzalkind