NatLabRockies/ROSCO #221: Where is "Wind/145m_11mps_ClassA1ETM.bts"?

lqw075, 2023-02-13:
Hello everyone,
I can't found "145m_11mps_ClassA1ETM.bts" in "ROSCO-2.5.0\ROSCO-2.5.0\Test_Cases\IEA-15-240-RWT-UMaineSemi\Wind".
I wonder where it is or how it is generated.
Thank you !!!

dzalkind, 2023-02-13:
Hi,

It's generated here: https://github.com/NREL/ROSCO/blob/main/Examples/08_run_turbsim.py. Though, the path may change.

Best, Dan

closed 2023-03-17 by dzalkind

lqw075, 2023-04-26:
I got this error when I ran "https://github.com/NREL/ROSCO/blob/main/Examples/08_run_turbsim.py. " with Python:
" ModuleNotFoundError: No module named 'ROSCO_toolbox' "
I would like to know the detailed operation steps or notes to generate turbulent wind smoothly.
Thank you!!!

dzalkind, 2023-04-26:
It looks like ROSCO was not installed correctly, if that is your error. 

Please try the [steps detailed here](https://rosco.readthedocs.io/en/latest/source/install.html) and let me know if you run into any issues with that.

Best, Dan

lqw075, 2023-04-27:
> It looks like ROSCO was not installed correctly, if that is your error.
> 
> Please try the [steps detailed here](https://rosco.readthedocs.io/en/latest/source/install.html) and let me know if you run into any issues with that.
> 
> Best, Dan

Thank you for your reply！This error occurred when I tried the steps：
: conda config --add channels conda-forge
conda create -y --name rosco-env python=3.7
conda activate rosco-env
cd <desired_folder>
conda install -y ROSCO
  File "<ipython-input-15-4c9bbad9b4f0>", line 1
    conda config --add channels conda-forge
               ^
SyntaxError: invalid syntax
I just changed python=3.8 to python=3.7,because my version is 3.7.

This error occurred when I run the code：
: git clone https://github.com/NREL/ROSCO.git
cd ROSCO
python setup.py install
  File "<ipython-input-17-6d4a8e605009>", line 1
    git clone https://github.com/NREL/ROSCO.git
            ^
SyntaxError: invalid syntax
I want to know how to solve this problem.
Thank you !!!

dzalkind, 2023-04-27:
You should run `git clone` from the command line.  You should probably also use python 3.8 (or higher).  This version of python will only be installed to your `rosco-env` conda environment, and it will not impact the python you have already installed on your machine.

lqw075, 2023-06-18:
> You should run `git clone` from the command line. You should probably also use python 3.8 (or higher). This version of python will only be installed to your `rosco-env` conda environment, and it will not impact the python you have already installed on your machine.

Thanks! This error occurs when I run the code:
runfile('E:/OPENFAST/3.1.0/_ExampleData/IEA-15-240-RWT/Wind/turbsim.py', wdir='E:/OPENFAST/3.1.0/_ExampleData/IEA-15-240-RWT/Wind')
Using E:/OPENFAST/3.1.0/ROSCO-2.5.0/Test_Cases/Wind/90m_12mps_twr.inp to run OpenFAST simulation
Running OpenFAST simulation for E:/OPENFAST/3.1.0/ROSCO-2.5.0/Test_Cases/Wind/90m_12mps_twr.inp through the ROSCO toolbox...
Traceback (most recent call last):

  File "E:\OPENFAST\3.1.0\_ExampleData\IEA-15-240-RWT\Wind\turbsim.py", line 27, in <module>
    run_openfast(wind_directory, fastcall='turbsim',

  File "E:\ProgramData\Anaconda3\lib\site-packages\ROSCO_toolbox\utilities.py", line 525, in run_openfast
    subprocess.run([fastcall, os.path.join(fastfile)], check=True, cwd=cwd)

  File "E:\ProgramData\Anaconda3\lib\subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:

  File "E:\ProgramData\Anaconda3\lib\site-packages\spyder_kernels\customize\spydercustomize.py", line 105, in __init__
    super(SubprocessPopen, self).__init__(*args, **kwargs)

  File "E:\ProgramData\Anaconda3\lib\subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,

  File "E:\ProgramData\Anaconda3\lib\subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,

FileNotFoundError: [WinError 2] The system cannot find the specified file.

This is my source code:

----------- 08_run_turbsim --------------
Run TurbSim to create wind field binary
-------------------------------------
In this example:
  - Leverage the run_openfast functionality to compile a turbsim binary
'''

# ROSCO toolbox modules 
from ROSCO_toolbox.utilities import run_openfast
import os

this_dir = os.path.dirname(os.path.abspath(__file__))

# Define openfast output filenames
wind_directory = os.path.join(this_dir,'E:/OPENFAST/3.1.0/ROSCO-2.5.0/Test_Cases/Wind/')
turbsim_infile = '145m_11mps_ClassA1ETM.inp'

run_openfast(wind_directory, fastcall='turbsim',
             fastfile=os.path.join(wind_directory, turbsim_infile), chdir=False)

print('test')
I want to know how to solve this problem.
Thank you !!!

dzalkind, 2023-06-20:
You need to define the full path to the turbsim executable on your computer in `fastcall='<path_to_turbsim.exe'`.

lqw075, 2023-07-01:
> You need to define the full path to the turbsim executable on your computer in `fastcall='<path_to_turbsim.exe'`.

Thanks ! It works.