NatLabRockies/ROSCO #222: ModuleNotFoundError: No module named 'wisdem'

hugocastroo, 2023-02-20:
when trying to install the ROSCO Toolbox or the complete ROSCO installation, getting the following error when trying to use the tune_ROSCO.py script.

wisdem package is correctly installed when running command (conda install -y wisdem), still getting error.

(rosco-env) C:\Users\huggo\ROSCO\Tune_Cases>tune_ROSCO.py
Using ofTools in ROSCO_toolbox...
Traceback (most recent call last):
  File "C:\Users\huggo\ROSCO\Tune_Cases\tune_ROSCO.py", line 26, in <module>
    from ROSCO_toolbox.inputs.validation import load_rosco_yaml
  File "C:\Users\huggo\AppData\Local\Programs\Python\Python39\lib\site-packages\rosco-2.6.0-py3.9.egg\ROSCO_toolbox\inputs\validation.py", line 2, in <module>
    from wisdem.inputs import validate_with_defaults
ModuleNotFoundError: No module named 'wisdem'

dzalkind, 2023-02-20:
Have you installed wisdem?  Usually `conda install widem` will work.  A version higher than 3.5 is required.

hugocastroo, 2023-02-21:
Yes I installed it, I have version WISDEM 3.8. Anyway, there are some compatibility problems with my PC. I tried it with my PC from work, and it is working, I did precisely the same steps, and the environment looks the same with the same modules installed, but for some reason, it does not work on my personal PC.
Thanks for your advice

closed 2023-02-21 by hugocastroo