NatLabRockies/ROSCO #224: Calling libdiscon.dll in Matlab using callib, loadlibrary and unloadlibrary functions

hugocastroo, 2023-02-22:
Hello,

I am trying to use different bladed-style controllers in Matlab for simulations. I was able to do some simulations with a c based dll. In order to call the controller in Matlab I was using a header file containing just the following content (just one line):

**void DISCON(float *avrSwap, int *aviFail, char *accInfile, char *avcOutname, char *avcMsg);**

I tried to load the ROSCO libdiscon.dll file using basically the same code but it is not working, Matlab is crashing and no error is being given. After a bit of looking for a solution, I found this comment in the NREL Forum:

_[Gerard.Ryan](https://forums.nrel.gov/u/Gerard.Ryan)
[Apr '22](https://forums.nrel.gov/t/calling-discon-libdiscon-dll-from-matlab-functions-loadlibrary-and-calllib/3152)
Dear Nikhar/Boni,
I have successfully coupled the Discon.dll with MATLAB using functions loadlibrary and calllib. **The procedure to implement this involved writing a header file in FORTRAN, which provides the definition and class of each variable being passed in and out of the DLL.** Using this implementation, I have developed an in house aeroelastic code (OxDyn) that closely matches OpenFAST for the NREL 5MW turbine as shown below:_

I also found an answer in Matlab forums, saying that when using the callib function, Matlab might crash if the license of the DLL is not the correct one.

Sadly, I have not been accepted into the NREL Forum yet and could not comment on this post to get more information about it. I could not find any example of a Fortran-based header file for something like this.

Do you have any example or hint about what can I try?

I was able to use exactly the same libdiscon.dll in OpenFAST with good results, so the DLL was compiled correctly and is working properly.

dzalkind, 2023-02-22:
Hi,

This sounds like a very useful feature, but we haven't had the bandwidth to pursue it.  Here is a comment I made on a PR a while back: https://github.com/NREL/ROSCO/pull/148#issuecomment-1168881319.  That is where I would start.

I hope this helps.  If you find a nice solution, we would be interested in supporting it for other users.

Best, Dan

hugocastroo, 2023-02-22:
Thanks for your feedback Dan, In case I got something, I will let you know!

Hugo

closed 2023-02-22 by hugocastroo