NatLabRockies/ROSCO #226: Does ROSCO 2.4.1 not work with .bts type wind files?

SYZ622, 2023-02-24:
When I run ROSCO again after changing the wind file to .bts, I get the following error
`Error reported by S-function 'FAST_SFunc' in 'ROSCO1/FAST Nonlinear Wind Turbine/S-Function':
FAST_Solution:FAST_AdvanceStates:SolveOption2c_Inp2AD_SrvD:InflowWind_CalcOutput:CalcOutput:IfW_FFWind_CalcOutput [position=(NaN, NaN, NaN)  in wind-file
coordinates]: Error: FF wind array was exhausted at 0.575 seconds (trying to access data at NaN seconds).`

SYZ622, 2023-02-25:
The Test_Cases that come with ROSCO use steady wind, but after I changed the incoming wind input file in Openfast to a wind file generated by Turbsim, I got the above error. Also, I would like to ask how to determine the version of ROSCO and the corresponding version of Openfast?

dzalkind, 2023-02-27:
They can be found in the print statements when OpenFAST and ROSCO are run.  Or if you installed via conda, you can `conda list | grep openfast` or `| grep rosco`. 

I suspect there may be an issue with your `bts` file if it runs okay in a steady wind.

closed 2023-03-17 by dzalkind