NatLabRockies/ROSCO #231: ROSCO Simulink runFAST.m error

void1981, 2023-03-11:
Hi everyone,
I am trying to run ROSCO Matlab toolbox as it is I have not changed anything in the `runFAST.m`,  except `FAST_SFuncDir` and `FAST_directory` values. trying it with the same example `5MW_Land_Simulink.fst`. 
Using ROSCO 2.5.0 and OpenFAST 3.1.0.
I am always getting these errors and warnings. here is Matlab log.
```
Warning: There are more headers in the table than columns. Ignoring the last 4 column(s).
WARNING: no outputs found in OutList

 **************************************************************************************************
 OpenFAST

 Copyright (C) 2022 National Renewable Energy Laboratory
 Copyright (C) 2022 Envision Energy USA LTD

 This program is licensed under Apache License Version 2.0 and comes with ABSOLUTELY NO WARRANTY.
 See the "LICENSE" file distributed with this software for details.
 **************************************************************************************************

 OpenFAST-v3.1.0
 Compile Info:
  - Compiler: Intel(R) Fortran Compiler 1900
  - Architecture: 64 bit
  - Precision: single
  - OpenMP: No
  - Date: Mar 08 2022
  - Time: 11:41:34
 Execution Info:
  - Date: 03/11/2023
  - Time: 15:38:28+0100

 OpenFAST input file heading:
     FAST Certification Test #18: NREL 5.0 MW Baseline Wind Turbine (Onshore)

 Running ElastoDyn.
 Nodal outputs section of ElastoDyn input file not found or improperly formatted.
 Running AeroDyn.
 AD15 Nodal Outputs: Nodal output section of AeroDyn input file not found or improperly formatted.
 Skipping nodal outputs.
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 3, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 4, blade 1)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 3, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 4, blade 2)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 1, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 2, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 3, blade 3)
 Warning: Turning off Unsteady Aerodynamics because UA parameters are not included in airfoil
 (airfoil has likely has constant polars). (node 4, blade 3)
 Running InflowWind.
 Running ServoDyn.

 FAST_InitializeAll:SrvD_Init:ValidatePrimaryData:Airfoil control is not commanded from Simulink
 model.
 ValidatePrimaryData:Cable control is not commanded from Simulink model.

Warning: Output port 1 of 'ROSCO/Filter WindSpeed' is not
connected. 
Warning: Input port 1 of 'ROSCO/BldPitch Controller/Mux' is not
connected. 
Warning: Input port 5 of 'ROSCO/BldPitch Controller/Mux' is not
connected. 
Warning: Output port 1 of 'ROSCO/BldPitch
Controller/Integral/Minimum Pitch' is not connected. 
Warning: Input port 2 of 'ROSCO/Cyclic Pitch controller/Sum2' is
not connected. 
Warning: Input port 3 of 'ROSCO/Cyclic Pitch controller/1P Cyclic
Pitch Controller1/saturation (limit schedule)1' is not connected. 
Warning: Input port 1 of 'ROSCO/Cyclic Pitch controller/2P Cyclic
Pitch Controller2/MBC Out' is not connected. 
Warning: Input port 3 of 'ROSCO/Cyclic Pitch controller/2P Cyclic
Pitch Controller2/saturation (limit schedule)1' is not connected. 
Warning: Output port 1 of 'ROSCO/PitchSaturation/Integral' is not
connected. 
Warning: Input port 2 of 'ROSCO/PitchSaturation/Sum' is not
connected. 
Warning: Output port 2 of 'ROSCO/PitchSaturation/Variable
Saturator1' is not connected. 
Warning: Output port 1 of 'ROSCO/SetpointSmoother/MinPitch' is not
connected. 
Warning: Unconnected input line found on 'ROSCO/Cyclic Pitch
controller/1P Cyclic Pitch Controller1/1P IPCDQ Filtering1' (input
port: 2) 

  OpenFAST completed.

Error using runFAST (line 59)
Error due to multiple causes.
Caused by:
    Error using runFAST (line 59)
    Error in port widths or dimensions. Output port 1 of
    'ROSCO/FAST Nonlinear Wind Turbine/Mux1' has 51 elements.
    This port does not accept the dimensions (or orientation)
    specified by the output signal.
    Error using runFAST (line 59)
    Error in port widths or dimensions. Input port 1 of
    'ROSCO/FAST Nonlinear Wind Turbine/S-Function' is a one
    dimensional vector with 94 elements.
```
Would you suggest any solution? Thank you

dzalkind, 2023-03-13:
Hi, please review earlier discussions (https://github.com/NREL/ROSCO/issues/172) and use the [OpenFAST Simulink example](https://github.com/OpenFAST/openfast/tree/main/glue-codes/simulink/examples) to get started with the correct port sizes.

void1981, 2023-03-13:
> Hi, please review earlier discussions (#172) and use the [OpenFAST Simulink example](https://github.com/OpenFAST/openfast/tree/main/glue-codes/simulink/examples) to get started with the correct port sizes.

Thank you for your reply. The Discussion (#172) discusses a different issue. (no outputs found in OutList)

[OpenFAST Simulink example](https://github.com/OpenFAST/openfast/tree/main/glue-codes/simulink/examples) works fine, but What interests me is ROSCO Simulink [ROSCO.mdl](https://github.com/NREL/ROSCO/blob/main/Matlab_Toolbox/Simulink/ROSCO.mdl). Shouldn't it be working?

wind2021, 2023-03-15:
Hello, I tried this setting as the picture below to solve this issue. 
[image]

dzalkind, 2023-03-15:
Hi, 

The Simulink model should work, but it depends on the model now that additional sensors/actuators are included in OpenFAST.  It's challenging to maintain the ROSCO controller in two coding environments (Simulink, Fortran).  The long-term goal is to use the Fortran library in Simulink.  The short-term goal may be to get a version running and stop supporting it after the next release.  As always, support from the community is welcome.

Best, Dan

void1981, 2023-03-17:
> Hello, I tried this setting as the picture below to solve this issue. 
> [image]
> 

That's actually solve the issue. Thanks alot

closed 2023-03-17 by void1981