NatLabRockies/ROSCO #238: Open loop yaw rate control

ymh-s, 2023-05-07:
Hi,

I'm trying to use the open loop control in ROSCO to control the nacelle yaw direction, but it doesn't seem to be working. For example the 14_open_loop_control.py case (the open loop torque and pitch control seem to be working properly).

I'm using OpenFAST 3.4.1 and ROSCO 2.8.0.

Thanks
Best regards

dzalkind, 2023-05-08:
Thanks for bringing this to my attention!  You are correct.  A fix is available in #239.  I will release a v2.8.1 in the coming days.

Best, Dan

dzalkind, 2023-07-19:
Hi, sorry for the long delay, but I don't think we'll be releasing a 2.8.1 anytime soon.  Looks like we'll release 2.9.0 with some new features.  This bug fix can be used with the latest `develop` branch.

closed 2023-07-19 by dzalkind