NatLabRockies/ROSCO #242: Compatibility with OpenFAST v3.5.0

Yuksel-Rudy, 2023-05-24:
Some of the input files in OpenFAST files in Test_Cases directory must be updated to match with the latest version of OpenFAST v3.5.0. For instance, IEA-15-240-RWT-UMaineSemi_InflowFile.dat must be updated as follows to include the new LiDAR capability:

on line 8, this has to be added:
```
False   			   VelInterpCubic - Use cubic interpolation for velocity in time (false=linear, true=cubic) [Used with WindType=2,3,4,5,7]
```

on line 51 to 63, LiDAR parameters must be added:
```
================== LIDAR Parameters ===========================================================================
          0   SensorType          - Switch for lidar configuration (0 = None, 1 = Single Point Beam(s), 2 = Continuous, 3 = Pulsed)
          0   NumPulseGate        - Number of lidar measurement gates (used when SensorType = 3)
         30   PulseSpacing        - Distance between range gates (m) (used when SensorType = 3)
          0   NumBeam             - Number of lidar measurement beams (0-5)(used when SensorType = 1)
       -200   FocalDistanceX      - Focal distance co-ordinates of the lidar beam in the x direction (relative to hub height) (only first coordinate used for SensorType 2 and 3) (m)
          0   FocalDistanceY      - Focal distance co-ordinates of the lidar beam in the y direction (relative to hub height) (only first coordinate used for SensorType 2 and 3) (m)
          0   FocalDistanceZ      - Focal distance co-ordinates of the lidar beam in the z direction (relative to hub height) (only first coordinate used for SensorType 2 and 3) (m)
0.0 0.0 0.0   RotorApexOffsetPos  - Offset of the lidar from hub height (m)
         17   URefLid             - Reference average wind speed for the lidar[m/s]
       0.25   MeasurementInterval - Time between each measurement [s]
      False   LidRadialVel        - TRUE => return radial component, FALSE => return 'x' direction estimate
          1   ConsiderHubMotion   - Flag whether to consider the hub motion's impact on Lidar measurements
```

dzalkind, 2023-05-25:
Hi Rudy,

Yep, we haven't updated:
- [Test_Cases](https://github.com/NREL/ROSCO/tree/main/Test_Cases)
- [FAST_Reader](https://github.com/NREL/ROSCO/blob/main/ROSCO_toolbox/ofTools/fast_io/FAST_reader.py)
- [FAST_Writer](https://github.com/NREL/ROSCO/blob/main/ROSCO_toolbox/ofTools/fast_io/FAST_writer.py)
- [CI_Requirement](https://github.com/NREL/ROSCO/blob/e3b7db779ad9e7bea5dea692e92f52b10116cf02/.github/workflows/CI_rosco-pytools.yml#L132)

for OpenFAST v3.5.0.  If you're feeling up to it, I'd be happy to let you try and update them.  I will be out for about a week starting next week and don't plan on updating these until after I return.

Best, Dan

dzalkind, 2023-07-19:
This was addressed in #246. Thanks again for your help, Rudy!

closed 2023-07-19 by dzalkind