NatLabRockies/ROSCO #248: ROSCO with repositioning PID nacelle-yaw controller

Dwivedi07, 2023-06-14:
Hello there!
I wanted to perform the repositioning analysis for IEA-15-240-RWT-UMaineSemi by adding a PID nacelle yaw closed-loop controller besides the ROSCO blade pitch and torque controller. I have performed the same analysis for 5MW by editing the .sln file in visual studio by adding closed loop nacelle yaw controller and bulid a new .dll file as output which I input to servo.dat. I want to do the same with 15MW and ROSCO controller but I am not sure how and where should I perform changes to implement nacelle yaw control to 15MW turbine.
Kind request to help me with this.
Thanks a lot

dzalkind, 2023-06-14:
Hi,

You can edit the existing yaw controller here: https://github.com/NREL/ROSCO/blob/e3b7db779ad9e7bea5dea692e92f52b10116cf02/ROSCO/src/Controllers.f90#L267 

Instructions for compiling ROSCO can be found here: https://rosco.readthedocs.io/en/latest/source/install.html#compile-using-cmake

Best, Dan

Dwivedi07, 2023-06-14:
Hello Dan,
Thanks for your reply;
So I can add my Nacelle Yaw controller here and then I can compile the ROSCO by doing cmake going in ROSCO/ROSCO/build directory, which will generate libdiscon.so in /ROSCO/install/lib directory.(as i am using conda configured rosco-env in Linux). Later I can input this .so file to my servo.dat , right? for simulation?
Thanks a lot

dzalkind, 2023-06-14:
That is correct. I'm converting this into a discussion since it's not really a software issue.