NatLabRockies/ROSCO #259: regarding FAST_reader.py

iman-tepsco, 2023-08-24:
I have two questions regarding _FAST_reader.py_. 

If the platform has more than one body, say two, and the model incorporates coupling between the bodies, the values for _NBody_ and _NBodyMod_ in HydroDyn will be two and one, respectively. In this scenario, _AddF0_ is a 12X1 vector (rather than a 6X1), while _AddCLin_, _AddBLin_, and _AddBQuad_ are 12X12 matrices (rather than 6X6). 
The dimensions of _AddF0_ are correctly understood by _FAST_reader.py_, but it appears to fall short of recognizing the dimensions of _AddCLin_, _AddBLin_, and _AddBQuad_, which may result in an error for such platforms. 

According to lines no. 1684-1693 of  _FAST_reader.py_: 

`

        if self.fst_vt['HydroDyn']['NBodyMod'] == 1:
            self.fst_vt['HydroDyn']['AddF0']         = [float(f.readline().strip().split()[0]) for i in range(6*NBody)]
        elif self.fst_vt['HydroDyn']['NBodyMod'] > 1:
            self.fst_vt['HydroDyn']['AddF0']         = [[float(idx) for idx in f.readline().strip().split()[:NBody]] for i in range(6)]
        else:
            raise Exception("Invalid value for fst_vt['HydroDyn']['NBodyMod']")

        self.fst_vt['HydroDyn']['AddCLin']       = np.array([[float(idx) for idx in f.readline().strip().split()[:6]] for i in range(6)])
        self.fst_vt['HydroDyn']['AddBLin']       = np.array([[float(idx) for idx in f.readline().strip().split()[:6]] for i in range(6)])
        self.fst_vt['HydroDyn']['AddBQuad']      = np.array([[float(idx) for idx in f.readline().strip().split()[:6]] for i in range(6)])`

I believe the issue is resolved if it is changed to the following: 

`

        if self.fst_vt['HydroDyn']['NBodyMod'] == 1:
            self.fst_vt['HydroDyn']['AddF0']         = [float(f.readline().strip().split()[0]) for i in range(6*NBody)]
            self.fst_vt['HydroDyn']['AddCLin']       = np.array([[float(idx) for idx in f.readline().strip().split()[:6*NBody]] for i in range(6*NBody)])
            self.fst_vt['HydroDyn']['AddBLin']       = np.array([[float(idx) for idx in f.readline().strip().split()[:6*NBody]] for i in range(6*NBody)])
            self.fst_vt['HydroDyn']['AddBQuad']      = np.array([[float(idx) for idx in f.readline().strip().split()[:6*NBody]] for i in range(6*NBody)])
        elif self.fst_vt['HydroDyn']['NBodyMod'] > 1:
            self.fst_vt['HydroDyn']['AddF0']         = [[float(idx) for idx in f.readline().strip().split()[:NBody]] for i in range(6)]
            self.fst_vt['HydroDyn']['AddCLin']       = np.array([[float(idx) for idx in f.readline().strip().split()[:6]] for i in range(6)])
            self.fst_vt['HydroDyn']['AddBLin']       = np.array([[float(idx) for idx in f.readline().strip().split()[:6]] for i in range(6)])
            self.fst_vt['HydroDyn']['AddBQuad']      = np.array([[float(idx) for idx in f.readline().strip().split()[:6]] for i in range(6)])
        else:
            raise Exception("Invalid value for fst_vt['HydroDyn']['NBodyMod']")`

Based on the foregoing, I have the following questions:
1. Is my comprehension correct? 
2. Is there anything we need to adjust in other files that use _self.fst_vt_? 
(For instance, is _self.fst_vt['HydroDyn']['AddCLin']_ regarded a 6X6 matrix in other files.)

Thank you very much for your time in advance. 

Iman

dzalkind, 2023-08-24:
Hi Iman,

I think your interpretation is correct. 

More information about the HydroDyn inputs can be found here: https://openfast.readthedocs.io/en/main/source/user/hydrodyn/input_files.html 

I've been wrestling with this same part of the FAST_reader/writer as well over [in this repo](https://github.com/dzalkind/WEIS/blob/411e43feb75640512b92bb8b8935b1cabdb0ba4b/weis/aeroelasticse/FAST_reader.py#L1765), but it may have not made it's way into ROSCO yet.  Our approaches look similar.

Best, Dna

iman-tepsco, 2023-09-01:
Hi Dna, 

Thank you for your reply. 

So my understanding appears to be logical. 

I have not yet tracked the program to see how _self.fst_vt['HydroDyn']['AddCLin']_, self.fst_vt['HydroDyn']['AddBLin'], and _self.fst_vt['HydroDyn']['AddBQuad']_ are utilized. 
I only need to know if their dimensions are calculated or if they are assumed to be 6X6 by default. 

I would appreciate your response in advance. 

Iman

dzalkind, 2023-09-01:
Hi Iman,

If more than one potential body is being modeled, it is not 6x6. It is like your proposed code suggests.

Best, Dan

dzalkind, 2024-11-22:
In OpenFAST 4.0, these file I/O routines will be handled by the `openfast_io` library

closed 2024-11-22 by dzalkind