NatLabRockies/ROSCO #268: Running ROSCO on Simulink

EhsanASL92, 2023-10-17:
Hi,
I am trying to run ROSCO on Simulink. First, I added the ROSCO_toolbox to set the path in Matlab.
Then, I changed the directory name for the SFunction and FAST directory.
fast.FAST_SFuncDir     = 'D:\OpenFASTbyEhsan\openfast-3.1.0\glue-codes\simulink\src';  %%%% NEED FOR SIMULINK
fast.FAST_InputFile    = '5MW_Land_BD_DLL_WTurb.fst';   % FAST input file (ext=.fst)
fast.FAST_directory    = 'D:\OpenFASTbyEhsan\openfast-3.1.0\reg_tests\r-test\glue-codes\openfast\5MW_Land_BD_DLL_WTurb';   % Path to fst directory files
 I tested it with both FAST v8 and OpenFAST v3.1.0
But I see this error :
Error in ReadWrite_FAST (line 14)
FP = FAST2Matlab(fullfile(fast.FAST_directory,fast.FAST_InputFile),2); %FP are Fast Parameters, specify 2 lines of header (FAST 8)
Can you help me with running ROSCO on Matlab/Simulink?

dzalkind, 2023-10-17:
Hi,

We can't properly test these simulink examples, so we can't support them quite as well.  Can you place a breakpoint deeper into `ReadWrite_FAST` to understand the issue it has?

Dan

EhsanASL92, 2023-10-17:
Thanks for your reply @dzalkind, 
In fact, I just tried to run the ROSCO controller. Based on what I see on the running guide, it cannot be hard to run.
I just defined some directories and then tried to use runFAST.m.
This is the first step for running ROSCO, but do not know why I see such an error!
I have attached the required files here.

[ReadWrite_FAST.zip](https://github.com/NREL/ROSCO/files/12936817/ReadWrite_FAST.zip)
[runFAST.zip](https://github.com/NREL/ROSCO/files/12936821/runFAST.zip)

dzalkind, 2023-10-17:
Please post the error that `ReadWrite_FAST` has.  Also, I encourage you to find a solution here because we cannot support Simulink issues at this time.

EhsanASL92, 2023-10-17:
Here is the error

**Unrecognized function or variable 'FAST2Matlab'.

Error in ReadWrite_FAST (line 14)
FP = FAST2Matlab(fullfile(fast.FAST_directory,fast.FAST_InputFile),2); %FP are Fast Parameters, specify 2 lines of header (FAST 8)

Error in runFAST (line 29)
[Param,Cx] = ReadWrite_FAST(fast);**

I check both github and also the NREL forum, but cannot get a good response for that yet!

dzalkind, 2023-10-17:
Have you tried searching for that function and adding it to your matlab path?

dzalkind, 2023-10-17:
Please see the documentation here: https://github.com/NREL/ROSCO/tree/main/Matlab_Toolbox 

You need the OpenFAST matlab-toolbox.

EhsanASL92, 2023-10-17:
Yes, I have already run the OpenFAST v3.1.0 and it is running correctly.
I also added all the necessary paths to my environment.

Liamkk, 2023-10-25:
I have encountered the same problem as you. Make sure the Matlab toolbox folder path is added to the Matlab path.

lweidsab, 2024-10-31:
Hi I am also encountered the same problem, while such error given.  The Matlab-Toolbox folder was added, and the fast_dir is:
fast.FAST_SFuncDir     = 'D:/openfast-main/glue-codes/simulink/src';  %%%% NEED FOR SIMULINK
fast.FAST_InputFile    = '5MW_Land_Simulink.fst';   % FAST input file (ext=.fst)
fast.FAST_directory    = 'D:/ROSCO/Examples/Test_Cases/5MW_Land_Simulink';   % Path to fst directory files

 
[image]

dzalkind, 2024-12-06:
Re-directing this issue to #263

closed 2024-12-06 by dzalkind