NatLabRockies/ROSCO #279: Running Problem using OpenFAST-Simulink

yugao0912, 2023-11-08:
Hello,

I was trying to use OpenFAST_Simulink function to run some examples. I just modified the paths of relevant files including:

[this_dir,~,~] = fileparts(mfilename('fullpath'));
fast.FAST_SFuncDir     = 'D:\OpenFAST\Op_Simulink\F2Simulink';  %%%% NEED FOR SIMULINK
fast.FAST_InputFile    = '5MW_Land_Simulink.fst';   % FAST input file (ext=.fst)
fast.FAST_directory    = 'D:\OpenFAST\Op_Simulink\ROSCO-main\Test_Cases\5MW_Land_Simulink';   % Path to fst directory files

The version of Rosco is 2.8.0. I have added all the necessary paths in matlab, however, there are some mistakes when running the simulation. Please see below:

Error using Pre_LoadRotPerf
Error loading file

Error in ReadWrite_FAST (line 47)
Cx                      = Pre_LoadRotPerf(fullfile(fast.FAST_directory,PerfFileName(2:end-1)));

Error in runFAST (line 27)
[Param,Cx] = ReadWrite_FAST(fast);

Could you please help me with this?

Thank you.

Best wishes,
Yu Gao

dzalkind, 2023-11-08:
Please see other issues and discussions about the matlab-toolbox, like this one: https://github.com/NREL/ROSCO/discussions/278#discussioncomment-7489802

Best, Dan

yugao0912, 2023-11-08:
> Please see other issues and discussions about the matlab-toolbox, like this one: [#278 (comment)](https://github.com/NREL/ROSCO/discussions/278#discussioncomment-7489802)
> 
> Best, Dan

Hello Dan,

I am sorry. I have checked the link you sent but I still have no idea about this problem. Maybe I just missed some information.

Best wishes,
Yu

dzalkind, 2023-11-08:
Can you set a breakpoint and see where the error occurs?

yugao0912, 2023-11-08:
> Can you set a breakpoint and see where the error occurs?

Hello,

[image]

The error occurs at line 27:

[Param,Cx] = ReadWrite_FAST(fast);

Best,
Yu

EhsanASL92, 2023-11-09:
> > Can you set a breakpoint and see where the error occurs?
> 
> Hello,
> 
> [image]
> The error occurs at line 27:
> 
> [Param,Cx] = ReadWrite_FAST(fast);
> 
> Best, Yu

I have the same problem, still looking for a solution!

dzalkind, 2023-11-09:
Have you tried stepping inside that function to see what the issue is?

yugao0912, 2023-11-09:
> Have you tried stepping inside that function to see what the issue is?

Hello,

In the ReadWrite_FAST.m file, the error occurs at line 14:

[image]

I just changed the path in the runFAST.m file.

Best,
Yu

yugao0912, 2023-11-09:
> Have you tried stepping inside that function to see what the issue is?

The error is 

[image]

Melancho7ee, 2024-02-17:
Dear @yugao0912 ,

I met similar errors like you before:
```
Error using Pre_LoadRotPerf
Error loading file

Error in ReadWrite_FAST (line 47)
Cx = Pre_LoadRotPerf(fullfile(fast.FAST_directory,PerfFileName(2:end-1)));

Error in runFAST (line 27)
[Param,Cx] = ReadWrite_FAST(fast);
```
My setting is:
```
[this_dir,~,~] = fileparts(mfilename('fullpath'));

% Compile FAST for use with simulink & mex using openfast docs
fast.FAST_SFuncDir     = './Simulink';  %%%% NEED FOR SIMULINK
fast.FAST_InputFile    = '5MW_ITIBarge_DLL_WTurb_WavesIrr.fst';   % FAST input file (ext=.fst)
fast.FAST_directory    = './5MW_ITIBarge_DLL_WTurb_WavesIrr';   % Path to fst directory files
```
As for me, the final reason is that there is no "Cp_Ct_Cq.NREL5MW.txt" in the directory of "fast.FAST_directory". After I copied this txt file to the './5MW_ITIBarge_DLL_WTurb_WavesIrr' directory, the error just disappered and everything is okay.

I'm not sure if your problem is the same as mine. The above contents are for your reference.

Best regards,

dzalkind, 2024-12-06:
Re-directing this issue to the main Matlab/Simulink issue #263

closed 2024-12-06 by dzalkind