NatLabRockies/ROSCO #284: The controller doesn't work in SIMA software

ChaojunYan, 2023-11-21:
no matter the ROSCO controller version, it doesn't work in SIMA software, e.g., in rate wind speed,  the blade pitch is around 3.4°. I tried to modify the DISCON file, change the ROSCO version, and import the DISCON and .dll files from GitHub or FAST, but they are all failed, the blade pitch, and rotor speed all not the same as the documentation (IEA-15-240-RWT_tabular.xlsx). 
PS: If I fix the blade pitch to the right value corresponding to the wind speed, the rotor speed and thrust are all right, so I believe the blade foils are all right.

dzalkind, 2023-11-21:
Hi, 

Can you explain a bit more what your setup is like and what behavior you are expecting to see?  We are not familiar with the SIMA software and how ROSCO interacts with it.  I would also recommend contacting SIMA for support.  I would be happy to also be included so we can better support ROSCO with SIMA.  My email is dzalkind (at) nrel (dot) gov.

To debug your issue further, you can use a `LoggingLevel` of 3 in the DISCON input.  Please use the latest version of ROSCO (v2.8.0).  ROSCO will output a `.dbg3` file that outputs the `avrSWAP`.  The `avrSWAP` is part of the standard Bladed interface for controllers.  In `avrSWAP(48)` the blade pitch command is sent to the modeling software.  [Sensor signals from the modeling software are read here](https://github.com/NREL/ROSCO/blob/e3b7db779ad9e7bea5dea692e92f52b10116cf02/ROSCO/src/ReadSetParameters.f90#L39).  Verifying the `avrSWAP` entry for `GenSpeed` may also help.

I hope this helps.

Best, Dan

ChaojunYan, 2023-11-22:
Dear Dan,
Thanks for your time!
The attachments are my detailed question description and controller input files.
[sima controller question.zip](https://github.com/NREL/ROSCO/files/13433087/sima.controller.question.zip)

dzalkind, 2023-11-22:
For the floating turbine you are using, blade pitch is not 0 unless you disable peak shaving with `PS_Mode` of 0.  However, it is not recommended to disable peak shaving for this model.  [A more detailed description of this controller can be found here.](https://wes.copernicus.org/articles/7/53/2022/#:~:text=The%20Reference%20Open%2DSource%20Controller,better%20than%20existing%20reference%20controllers.)

Generally, it looks like the pitch angle command is being sent to SIMA without issue.  Reviewing the avrSWAP entries would confirm this.

dzalkind, 2023-11-22:
Also, I heard from a SIMA developer that the wind speed estimator doesn't always work with SIMA.  Try disabling the WSE with `WE_Mode` of 0.

dzalkind, 2023-11-22:
Alternatively, the `develop` branch of ROSCO should address this issue, but you will need to compile from source and update your DISCON inputs.

ChaojunYan, 2023-12-06:
Hy, I'm sorry bother you again, I tired to disabling the WSE with WE_Mode of 0, it still not works. I was wondering which controller version was employed in the NREL report (IEA Wind TCP Task 37 Definition of the IEA Wind 15-Megawatt Offshore Reference Wind Turbine Technical Report) to get this blade pitch controll results?
[image]
In the ROSCO v2.6 controller, the MINIMUM PITCH SATURATION in IN file is given as the curve below. As I understand, this curve is responsible for the minimum pitch angle in the analysis. Then when the wind speed is 10m/s, the pitch angle cannot go down to 0 deg. 
[image]
And I tried with ROSCO v2.8 still can not get the same blade pitch results as the report.
So, could you pls tell me which controller version could get that results, thanks a lot!

dzalkind, 2023-12-07:
What issue in these results are you trying to resolve?  The results in your last plot look as expected.  Perhaps the SIMA inputs have more turbulence?  The rotor average wind speed might highlight this difference.

ChaojunYan, 2023-12-08:
Thanks for your reply！
1. I was wondering the balde pitch should be 0 deg from wind speed 6.98 to 10.59 m/s according to the NREL report, why you mention blade pitch should be 1-2 deg at 10m/s?
2. My stationary rate wind of 10.59 m/s results are as below, the result in sima seems not that reguarly.
[image]
3. I used the same turbulence wind input file in sima and fast, the wind speed quite the same actually.
[image]
[image]

ChaojunYan, 2023-12-08:
Thanks for your reply!
1. My stationary rate wind of 10.59 m/s results are as below, the results still have little difference but acceptable, how can i improve this?
[image]
2. I used the same turbulence wind input file in sima and fast, the blade pitch controll not well, the wind speed quite the same actually.
[image]
[image]

dzalkind, 2023-12-12:
Are you using the same turbsim grid in both simulators?  Does SIMA use a full-field turbulence like OpenFAST or does it take the turbulence and apply it uniformly across the rotor disk?

You can use the `dbg2` files generated by ROSCO to debug this further.  They are generated when LoggingLevel = 2 in the DISCON.  You will see `WE_Vw` (the wind speed estimate) and `PC_MinPit` (the minimum pitch saturation limit) there.  `PC_MinPit` is a function of `WE_Vw` that you can fine-tune in the peak shaving section of the DISCON.  I suspect there are slight offsets in `WE_Vw` that you will need to account for to match the response.  The inputs to the WSE are in the `dbg` files.  They are `WE_*`.  Ideally, given the same inputs, the WSE should produce the same output (`WE_Vw`) regardless of the simulator.  However, if you disable the WSE with `WE_Mode` of 0, the WSE is just a filtered hub height wind speed.

ChaojunYan, 2023-12-13:
Thanks for your reply, I used the same turbsim grid in both simulators. I didn't find the notes for the turbulence field in Sima, but I think SIMA also use full-field turbulence like OpenFAST.
Following your suggestion, I ran two conditions.
For WE_Mode=2, the results of dbg2 files are as below, wind speed estimator not work in Sima indeed.
[image]
[image]
For WE_Mode=0, the results of dbg2 files are as below, there are slight offsets in WE_Vw indeed, what is the reason for this? Can you help me answer it. And why does PC_MinPit seem to not change over time in Sima, regardless of WE_ Is MODE 0 or 2.
[image]
[image]

dzalkind, 2023-12-13:
Can you confirm that the DISCON inputs are identical?  There should be a mapping from B to C in your plots.  There may be another setting interfering with the peak shaving.

ChaojunYan, 2023-12-14:
Yes, I used the same DISCON inputs

dzalkind, 2023-12-15:
This doesn't really make sense to me.  PC_MinPit is determined [here](https://github.com/NREL/ROSCO/blob/e3b7db779ad9e7bea5dea692e92f52b10116cf02/ROSCO/src/ControllerBlocks.f90#L374) and [here](https://github.com/NREL/ROSCO/blob/e3b7db779ad9e7bea5dea692e92f52b10116cf02/ROSCO/src/Controllers.f90#L77).

Could you share the exact DISCON.IN files used in each set up and the `.dbg2` files generated by each set up?

ChaojunYan, 2023-12-15:
Thanks for your reply, the attachment is my detail discon.in and .dbg2 files, and the .docx is my test between sima and fast using two rosco controll version, seems wind speed greater than rate speed, controll works differents.
[DISCON.IN & dbg2.zip](https://github.com/NREL/ROSCO/files/13681896/DISCON.IN.dbg2.zip)

dzalkind, 2023-12-15:
The dbg2 files are not included in this zip file.  Can you please include them for both OpenFAST and SIMA?

dzalkind, 2023-12-15:
In the meantime, please also try this file:
[DISCON-UMaineSemi.IN.txt](https://github.com/NREL/ROSCO/files/13687571/DISCON-UMaineSemi.IN.txt)

In your attachment, it looks like the issue with the minimum pitch is resolved.

ChaojunYan, 2023-12-16:
[fast.zip](https://github.com/NREL/ROSCO/files/13691578/fast.zip)
[sima.zip](https://github.com/NREL/ROSCO/files/13691579/sima.zip)
Thanks, the minimum pitch much imporved, but the still not very well when wind speed greater than rate speed